MarlinFirmware/Marlin #13549: SKR 1.3 as EFF instead of EFB

macasero, 2019-04-01:
I would like to know if there is a version board to set SKR 1.3 board to enable either

exturder01 - heater01 for FAN1
or
heater00 (heated bed) for FAN1

So I don´t have to use the always on fan connector for hotend and case fan

gloomyandy, 2019-04-01:
Assuming you only have one extruder then FAN1 is already configured in the standard pins file...
```C
//
// Heaters / Fans
//
#define HEATER_0_PIN       P2_07
#if HOTENDS == 1
  #define FAN1_PIN         P2_04
#else
  #define HEATER_1_PIN     P2_04
#endif
#define FAN_PIN            P2_03
#define HEATER_BED_PIN     P2_05
```
So all you need to enable say automatic operation of the hotend cooling fan is...
#define E0_AUTO_FAN_PIN FAN1_PIN
in your configuration file.

sjasonsmith, 2020-09-11:
Based on the commentary above this does not appear to be an actionable request at this time. Marlin currently has logic which will map unused heater outputs to fans. If issues exist in this functionality it makes sense to report those as new issues specific to the latest state of Marlin.

closed 2020-09-11 by sjasonsmith