MarlinFirmware/Marlin #13551: [BUG] Ender 3 wrong CHOPPER_TIMING at default config

ArturNadolski, 2019-04-01:
### Description

### Steps to Reproduce

Current deffinition of CHOPPER_TIMING at Ender 3 Configuration_adv.h files:

#define CHOPPER_TIMING CHOPPER_DEFAULT_12V

Ender 3 is using 24V for steppers so CHOPPER_TIMING need to be set CHOPPER_DEFAULT_24V

JosefKuchar, 2019-05-25:
Fixed now so it can be closed

boelle, 2019-07-17:
@ArturNadolski can you confirm its fixed and click close button below?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

hhoweson, 2019-08-04:
I just download the latest release and noticed that the default chopper timing is still 12v for the ender 3 config, are we sure this has been fixed?

ArturNadolski, 2019-08-05:
As I wrote at OP... Ender 3 is 24V so please set CHOPPER_DEFAULT_24V as default.

hhoweson, 2019-08-05:
I only meant to point out that the issue has been closed yet it does not seem to be fixed

InsanityAutomation, 2019-08-05:
Seems this was rebroken with a careless update last month. It had been fixed at one point.

InsanityAutomation, 2019-08-05:
I should note this is only valid for people upgrading the board to an SKR or something as thats only used for uart controlled drivers.

PR #13565 (2019-04-04): Change to CHOPPER_TIMING to 24V for Ender3
### Description

Ender 3 is a 24V machine. Fixes #13551.

> Current definition of CHOPPER_TIMING at Ender 3 Configuration_adv.h file:
> #define CHOPPER_TIMING CHOPPER_DEFAULT_12V
> Ender 3 is using 24V for steppers so CHOPPER_TIMING need to be set CHOPPER_DEFAULT_24V
PR #14851 (2019-08-07): Re-fix Ender 3 Chopper setting
Re-fix #13551