MarlinFirmware/Marlin #13555: [BUG] RADDS + RRD full graphic doesn't boot until client connects on USB

ej0rge, 2019-04-02:
### Description

A couple months ago I installed Marlin 2.0-bugfixes on a RADDS+Due to upgrade the controller on my Tronxy X5S. I realize I'm way late to the game with that board but i bought it so I don't see why i shouldn't try and use it.

 I'm still using the tronxy 12864 display which is a 12864 lcd oddly attached to an RRD Smart Controller 2004 board without the SD card or beeper installed. 

I know that when using the original red tronxy_melzi board marlin prefers to think of it as lcd_for_melzi but it has the dual cables like the RRDFGSC and works as one if you plug it into a RAMPS with a "smart adapter". Anyhow, I made my own adapter cable out of female-to-male jumpers and 5 position dupont shells glued together on the display end, whatever the other sizes were on the RADDS side, and this works ok. Some issues with the encoder but that may be configuration issue or cable length issue. I'll figure that out later. 

I removed the small fets (T1 and T2? I forget) from the DUE board to prevent it from being powered over USB, since I am using trinamic stepper drivers which need VMOT powered before VDD. 

When I power on the printer, the display is lit up but blank, 

I did look at the closed issue(s) about blank displays and the ramps, and i am pretty sure those were different issues. 

All the aux IO pins are held high in this state with the blank screen, which suggests to me that marlin hasn't finished booting, or maybe hasn't begun, or something? 

When I click 'connect' in Octoprint (USB already attached), the display springs to life and the AUX pins go to their low state, and Marlin is obviously running, and seems to work fine with no issues that i have run into.  

Initially, I did have printcounter enabled, but disabling it has made no difference. I've rebuilt and run m502 several times since disabling printcounter, making different changes to my configuration. 

I am building with Arduino 1.9.0-Beta but i have platformio working if someone thinks that is worth a shot. 

I'm using the "programming" port to connect the Due to my octopi server. Using the native port seemed like the natural thing at first but it didn't seem to work? I'm not sure if it's supposed to?

### Steps to Reproduce

Install marlin on radds/due, apply power

**Expected behavior:** [What you expect to happen]

Marlin should boot?

**Actual behavior:** [What actually happens]

Doesn't appear to boot.

[X5S.zip](https://github.com/MarlinFirmware/Marlin/files/3032271/X5S.zip)

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@ej0rge is this still a problem?

ej0rge, 2019-09-29:
> @ej0rge is this still a problem?

This printer has been down for months due to unrelated circumstances, but I will have some down time this week and will take some time to double-check the wiring and flash a current marlin 2.0.

InsanityAutomation, 2019-10-15:
This is an issue with a couple platforms, including STM32F4 (Btt SKR1.1Pro) and we need to determine if its bootloader related or not at some point.

boelle, 2019-11-06:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

boelle, 2019-12-07:
i guess we also need to look at this one?

boelle, 2019-12-24:
@InsanityAutomation wonder if this would be better suited for a milestone in the future?

boelle, 2020-01-08:
any news if this is a bootloader broblem?

ej0rge, 2020-01-09:
I am actually working on this now. I have my X5S powered up and working no worse than it was. 

I have to update my configuration for 2.0.1 and then i will install fresh firmware. If that doesn't change things, I will look into updating the bootloader(s) on my Due, which was an aliexpress purchase.

boelle, 2020-01-14:
remember to use bugfix 2.0.x and not 2.0.1
2.0.1 is just a frozen copy of bugfix 2.0.x and hence its not as new as bugfix 2.0.x

closed 2020-01-21 by boelle