MarlinFirmware/Marlin #13556: LCD Screen not working

DerAdministrator, 2019-04-02:
Hello there

I have a Tronxy X5S-2E (3D Printer), a Fysetc F6 V1.3 motherboard and the included LCD screen(Links unterneath). Now I have plugged everything together. The motherboard has power and the lcd screen is white. So I started with the installation of the LCD Firmware. I used the the ADVi3pp firmware and did everythink like in the guide. I've used "Win32 Disk Imager 1.0" to load the .img file to the micro sd card. After that i downloaded the marlin .hex file form the guid and uploaded that directly to the mainboard via "Xloader". The upload worked, the mainboard LED was flashing while uploading and Xloader showed "upload finished". So I pluged the micro sd card into the LCD screen and turned on the power. Still the same result, the LCD screen was just white.

My question now is, what did I do wrong and why is my LCD screen just white?

Thanks
Kevin

Links
[Tronxy X5S-2E](https://www.gearbest.com/3d-printers-3d-printer-kits/pp_009218826490.html?wid=1433363)

[Fysetc F6 V1.3 & LCD screen](https://de.aliexpress.com/item/FYSETC-F6-V1-3-ALLE-in-one-Mainboard-4-3-Touch-Screen-6-st-cke-TMC2100/32958588999.html?spm=a2g0s.8937460.0.0.2bb02e0e2rujbP)

[ADVi3pp](https://github.com/andrivet/ADVi3pp-Marlin/releases)

[.img file](https://github.com/andrivet/ADVi3pp-Marlin/releases/download/advi3%2B%2B3.0.2/ADVi3pp-LCD-3.0.2.img.zip)

[.hex file](https://github.com/andrivet/ADVi3pp-Marlin/releases/download/advi3%2B%2B3.0.2/ADVi3pp-Mainboard-3.0.2.hex)

[Win32 Disk Imager 1.0](https://sourceforge.net/projects/win32diskimager/)

[Xloader](http://xloader.russemotto.com/)

jmz52, 2019-04-02:
Display you have connects to board via COM port and has it's own proprietary firmware.
Problem you have is not related to Marlin firmware.
You should contact vendor and/or seller for help with this display.

7eggert, 2019-04-03:
From the instructions, I found:
"(You can't just use the USB to power the board, if you do that, the screen will be powerless and display white screen.)"

If you did apply +12V, did you check the direction of the power cables?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
Keep in mind this question is probably better suited to the branch of firmware you flashed with than here. Its fairly setup specific and not many here will know the particulars of it.