MarlinFirmware/Marlin #13558: Watchdog timer failing to trigger on Archim 2.0

marcio-ao, 2019-04-02:
On Archim 2.0, we have noticed some rare occasions where a printer locks up and going into thermal runaway without triggering a reboot. The printer's UI is frozen and unresponsive.

I suspect the underlying cause may be a corrupted USB flash drive or noise on the ribbon cable, but in any case, it seems like the watchdog timer should prevent the printer from locking up and entering thermal runaway. However, this has not been happening in all cases.

Looking through the code, I noticed that very few places that call `watchdog_reset()`, which is reassuring, but `Marlin/src/sd/Sd2Card.cpp` stands out as possibly a place where it should not be called. In the interest of safety, perhaps Sd2Card should call `thermalManager.manage_heater();` rather than `watchdog_reset()`.

Unfortunately, since we have USB_FLASH_DRIVE_SUPPORT enabled, that code isn't even reached, so something else must be causing the watchdog timer fail to activate on the Archim.  Any thoughts?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

marcio-ao, 2019-07-24:
Since there was some activity on this, let me add some more information. We had some additional instances where this was happening and I suspect that the watchdog timer is effective for code that runs in a infinite loop (such as "for(;;)"), but it is not effective in code that gets stuck in a recursive loop that happens to involve a call idle(). In the case where we had a watchdog failure, our UI was requesting a toolhead change, but changing the toolhead somehow causing the idle() function to be called before active_extruder was changed, which caused our UI to request another toolhead change, etc. This recursion was such that I presume only parts of the idle() function had a chance to execute.

One thing I noticed is that `watchdog_reset()` is called in [the code that reads the temperature](https://github.com/MarlinFirmware/Marlin/blob/1088846caef6f660d5af65e1725a6a6deeef16fd/Marlin/src/module/temperature.cpp#L1505). It seems like it should be reset in the code that compares the read temperature to the target temperature and turns the heaters on or off based on that measurement. What keeps your house from burning down is not measuring the temperature -- it is turning off the heater when the temperature is too hot.

boelle, 2019-09-24:
@marcio-ao i assume this is still a problem?

marcio-ao, 2019-09-25:
@boelle: We haven't encountered this issue in a while. Closing this ticket.

closed 2019-09-25 by marcio-ao