MarlinFirmware/Marlin #13560: Compilation issue STM32F407Discovery

ur5zeb, 2019-04-02:
Hi, I have a problem when compiling
How to fix? 

AssertionError: :
File "C:\users\work\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\work\.platformio\platforms\ststm32\builder\main.py", line 90:
target_elf = env.BuildProgram()
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\work\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\work\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 319:
SConscript(env.GetFrameworkScript(f), exports="env")
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\work\.platformio\platforms\ststm32\builder\frameworks\arduino.py", line 50:
"tools", "platformio-build.py"))
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\work\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\users\work\.platformio\packages\framework-arduinoststm32\tools\platformio-build.py", line 38:
assert isdir(CMSIS_DIR)
============================================================================================================================================ [ERROR] Took 16.98 seconds ============================================================================================================================================

===================================================================================================================================================== [SUMMARY] =====================================================================================================================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [ERROR]
Environment ARMED                       [SKIP]
Environment mks_robin                   [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
Environment linux_native                [SKIP]
=============================================================

Idorobots, 2019-04-05:
Looks to be your platformio setup. What version are you using? (You can find out by running `platformio --version` in the console.) Could you also provide your `Configuration.h` and `Configuration_adv.h` files for reference?

ur5zeb, 2019-04-05:
platformio --version 3.6.6  Files Configuration.h and Configuration_adv.h from the examples Marlin-bugfix-2.0.x\config\examples\STM32F4 without modifications
platformio.ini
`[platformio]
src_dir     = Marlin
build_dir   = .pioenvs
lib_dir     = .piolib
libdeps_dir = .piolibdeps
boards_dir  = buildroot/share/PlatformIO/boards
env_default = STM32F4`

After reinstalling platformio, 
I have the following errors
{
	"resource": "/3DPrinter/Marlin-bugfix-2.0.x/Marlin/src/HAL/HAL_STM32F4/HAL_spi_STM32F4.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'class SPIClass' has no member named 'dmaTransfer'; did you mean 'transfer'?",
	"startLineNumber": 132,
	"startColumn": 9,
	"endLineNumber": 132,
	"endColumn": 9
}

{
	"resource": "/3DPrinter/Marlin-bugfix-2.0.x/Marlin/src/HAL/HAL_STM32F4/HAL_spi_STM32F4.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'class SPIClass' has no member named 'dmaSend'",
	"startLineNumber": 166,
	"startColumn": 9,
	"endLineNumber": 166,
	"endColumn": 9
}

{
	"resource": "/3DPrinter/Marlin-bugfix-2.0.x/Marlin/src/HAL/HAL_STM32F4/HAL.h",
	"owner": "cpp",
	"severity": 8,
	"message": "#error \"Serial port 0 does not exist\"",
	"startLineNumber": 58,
	"startColumn": 4,
	"endLineNumber": 58,
	"endColumn": 4
}

And many others :(

boelle, 2019-07-17:
put Configuration.h and Configuration_adv.h and platformio.ini in a zipfile and attach

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle