MarlinFirmware/Marlin #13561: [BUG][Marlin-bugfix-0.x]When USART ORE trigger,it will always go to usart_irq in STM32F1

ElliotCui97, 2019-04-03:
When serial ORE IRQ trigger,it will always go to usart_irq in STM32F1 ,looks like the program is dead.
There is a universality problem in STM32F1,need to add some code to clean ORE flag.
there is my measures：
Add these code in .plaftormio\packages\framework-arduinoststm32@2.1.180219\STM32F1\system\libmaple\usart_private.h line 63:
    if(regs->SR & USART_SR_ORE){ //if ORE,clean ORE flag
        uint32_t tempreg;
        tempreg = regs->SR;
        tempreg = regs->DR;
    }

ElliotCui97, 2019-04-03:
a little problem,why not use DMA to recive and send USART message?

jmz52, 2019-04-03:
Please provide some details how this impacts Marlin operations and how this can be reproduced. 'Issue: Bug report' template has a good example of information needed.

Also please check if this problem still exists on latest version of PlatformIO packages - framework-arduinoststm32 you are using is a bit outdated.
Current versions are 3.10500.190327 for framework-arduinoststm32 and 1.10000.190228 for framework-arduinoststm32-maple.

ElliotCui97, 2019-04-03:
ststm32 framwork@4.3.1
framework-arduinoststm32@2.1.180219
the platform.ini define it use platform must be older than 4.4
#
# STM32F103RE
#
[env:STM32F1]
platform      = ststm32@<4.4.0
framework     = arduino
board = genericSTM32F103RE

jmz52, 2019-04-03:
Here is an instruction how to compile STM32F1 code using latest STSTM32 version.
https://github.com/MarlinFirmware/Marlin/issues/13075#issuecomment-478658051

ElliotCui97, 2019-04-03:
ok,i will try to use new STSTM32 version

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle