MarlinFirmware/Marlin #13563: [BUG] Extruder motor doesnt run if LIN_ADVANCE enabled with K == 0

xorza, 2019-04-03:
### Description

Enabling LIN_ADVANCE with k value == 0 on SKR V1.3 mb with TMC2130 SPI, makes extruder motor never rotate

### Steps to Reproduce

0. Use 2.0_bugfix branch
1. Use SKR V1.3 motherboard
2. Use TMC2130 SPI
3. Enable LIN_ADVANCE and set K to 0.0
```
#define LIN_ADVANCE
#if ENABLED(LIN_ADVANCE)
  #define LIN_ADVANCE_K 0.0
#endif
```

**Expected behavior:** Extruder motor works

**Actual behavior:** Extruder motor never spins (it has voltage on it though, it is hard to spin it with fingers). Disabling LIN_ADVANCE fixes problem.

#### Additional Information

https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...xorza:skr1.3_wanhao here are all my changes, but I'm sure that no other thing affects my problem except LIN_ADVANCE (I tried reverting all the changes I made)

I didnt try with any other motor driver, I have only TMC2130 SPI

kAdonis, 2019-04-03:
Duplicate of #13535 and others

`#define MINIMUM_STEPPER_PULSE 1` seems to fix the issue

this should really be standard for TMC drivers

xorza, 2019-04-03:
@kAdonis thanks for prompt reply! should I close this now?

closed 2019-04-03 by xorza