MarlinFirmware/Marlin #13564: [BUG] No beeps during filament change in latest bugfix-2.0.x

rmoravcik, 2019-04-03:
Hi all,

I'm using the latest bugfix-2.0.x and Marlin stopped generating beeps during filament change after commit 9a515cbd3268b30eb7a4dc9ea0f96f3a253d9063.

I was looking to the code, and it seems, that pause_mode variable from src/feature/pause.cpp is always set to PAUSE_MODE_PAUSE_PRINT and filament_change_beep() function will always exiting without beeps.

@thinkyhead Could you please have a look to it? It seems you were trying to fix beeps in some way.

Thank you,
Roman

bossman74, 2019-04-22:
Same here, also I don't know if this is new, but the hotend doesn't cool down afterwards.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@rmoravcik i assume this is still an issue?

rmoravcik, 2019-09-27:
@boelle Sorry for my response. I just tested the latest bugfix-2.0.x and it's still an issue.

boelle, 2019-10-21:
@rmoravcik i assume this is still an issue?

last time i had a filement sensor and had it enabled in firmware there was plenty of beebs

but i could test it again, i just need to find the switch again or print a new one

VanessaE, 2019-10-22:
This is indeed still an issue (referring to filament change as commanded by `M600`, not by a filament sensor).   bugfix-2.0.x, commit e604f767.

boelle, 2019-10-22:
so e604f76 broke it or was it the last commit where it worked?

VanessaE, 2019-10-22:
Neither -- I am *at* e604f76, and filament-change-beeping does not work here.  No idea when it broke.

rmoravcik, 2019-10-22:
As I stated in the report, it was broken by commit https://github.com/MarlinFirmware/Marlin/commit/9a515cbd3268b30eb7a4dc9ea0f96f3a253d9063

rmoravcik, 2019-11-07:
The problem was fixed in bugfix-2.0.x by #15783.

closed 2019-11-07 by rmoravcik