MarlinFirmware/Marlin #13567: [BUG] the pin assignment for the Rumba board don't corespond to neither of the schematics I have found for the BIQU Tango v1.0 and Rumba board which have identical pin outs. So I can't assign the correct pins for my TMC2130 drivers.

Pan369, 2019-04-03:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

psavva, 2019-04-06:
Please provide the details as per the issue template including config files

gloomyandy, 2019-04-06:
It would also help if you provided links to the "Two schematics" you have referred to.

Bob-the-Kuhn, 2019-04-08:
You might also try using the **M43 W I1** and **M43 T** utilities to identify the logical pin to physical pin mapping.

Bob-the-Kuhn, 2019-04-09:
This is what the **M43 W I1** and a 1K jumper to ground gave me for EXP3.

I just noticed that I have a MKS RUMBA+ V1.0 board which is definitly different than a RUMBA.  There are more pin headers and the thermal connector has been re-arranged.  So far the stepper sockets, endstops and htermal inputs all match the Marlin pins so I expect that EXP3 below is correct.

[image]

Pan369, 2019-04-11:
Hi there, everyone. Sorry for taking so long to reply. Was "off line" till just now. Here are the two schematics:
[Tango.V1.schema.pdf](https://github.com/MarlinFirmware/Marlin/files/3068178/Tango.V1.schema.pdf)
[image]. 
Meanwhile I put the Tango board aside, thinking I have smoked it by hastily connecting a three wire end stop... and tried a RAMPS 1.6 with Funduino, but can't get the Discount Full Graphics LCD to work with it so back to the Tango which looks to be ok. Gonna try the Marlin logical pin numbers Bob-the-Kuhn suggested and I'll get back and post my config. files. Thanx for all your responses!

Pan369, 2019-04-11:
[image]

Pan369, 2019-04-13:
YES! Thanx Bob! The Marlin logical pin numbers work on the Tango v1.1.

closed 2019-05-18 by thinkyhead

Aurawin, 2019-08-30:
I know this is an old thread but I think I've found another issue with the pinsRumba.h

#ifndef TEMP_0_PIN
  #if TEMP_SENSOR_0 == -1
    #define TEMP_0_PIN      6   // Analog Input (connector *K1* on RUMBA thermocouple ADD ON is used)
  #else
    #define TEMP_0_PIN     15   // Analog Input (default connector for thermistor *T0* on rumba board is used)
  #endif
#endif

#ifndef TEMP_1_PIN
  #if TEMP_SENSOR_1 == -1
    #define TEMP_1_PIN      5   // Analog Input (connector *K2* on RUMBA thermocouple ADD ON is used)
  #else
    #define TEMP_1_PIN     14   // Analog Input (default connector for thermistor *T1* on rumba board is used)
  #endif
#endif

The A9 and A10 analog inputs aren't mapped to the proper pins for the TANGO board.  The A9 pin is actually 10 (not 6) and the A10  is 9 (not 5).  I recall having to move over one of my PWM fans to match that of the FAN0 or FAN1.  So maybe that one can be brought over as well.

#define TEMP_0_PIN 9 // A9=9 A10 = 10
#define TEMP_SENSOR_0 -1 //11

I think the only real way to solve this problem is to break out the section into a Tango_pins.h file

thinkyhead, 2019-09-03:
> The A9 and A10 analog inputs aren't mapped to the proper pins for the TANGO board

If TANGO is different from RUMBA, then you are correct. It needs its own pins file. If it is mostly like RUMBA then it can just include the RUMBA pins file and apply mods.

Aurawin, 2019-10-08:
I have a tango board. FAN_PIN is pin 7 and FAN1_PIN is pin 8.  The printouts on the board do label these as Fan0 and Fan1.

This is different than the pins_rumba.h where Fan_PIN is pin 8 and Fan1_PIN 7

In the past I just swapped out the two fans at the board, but leaving this issue unreported can lead to overheating or thermal runaway for newbies.

So I recently I reverted the wires back to see if I can code my way out of this.

Attempt 1:
Configuration_adv.h
#define E0_AUTO_FAN_PIN 7
error: "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN.

Attempt 2:
Configuration_adv.h
#define FAN_PIN 8 
#define E0_AUTO_FAN_PIN 7
error: 
sketch\src\gcode\control\M42.cpp: In static member function 'static void GcodeSuite::M42()':
sketch\src\gcode\control\M42.cpp:62:9: error: duplicate case value

         case FAN1_PIN: thermalManager.fan_speed[1] = pin_status; break;

Attempt 3: 
pins_Rumba.h
#ifndef FAN_PIN
  #define FAN_PIN           8
#endif
#ifndef FAN1_PIN //  <-------------------------- I added this to stop the pins_Rumba.h from overwriting my value
  #define FAN1_PIN            7
#endif

Attempt 4:
Configuration_adv.h
#define FAN_PIN             8
#define FAN1_PIN            -1   // <---- This not possible with current version
#define E0_AUTO_FAN_PIN 7

pins_Rumba.h
#ifndef FAN_PIN
  #define FAN_PIN           7
#endif
#ifndef FAN1_PIN //  <-------------------------- I added this to stop the pins_Rumba.h from overwriting my value
  #define FAN1_PIN            8
#endif

As it stands I can't build Marlin with as is.  I need to add #ifndef FAN_1PIN clause so I can overwrite Configuration_adv.h with :
#define FAN_PIN             8
#define FAN1_PIN            -1   // <---- This not possible with current version
#define E0_AUTO_FAN_PIN 7

Is there any way we can have a pins_Tango.h board for a few of these changes?  If not, can we have the above change to add an ifndef for FAN1_PIN?

I have made sandbox changes but next time I overwrite, these changes will be lost.  Any thoughts?

Aurawin, 2019-10-24:
> YES! Thanx Bob! The Marlin logical pin numbers work on the Tango v1.0.
> That is to say that I got Z & Y steppers moving, love the silence!, but I get an over temp warning on E0 & X. I know the drill. Thanx again for all your responces!

In the off chance that you are still following this thread... What pins and what values did you use to solve the problem.

I'm creating a pins_Tango header to hold all the changes in pins.  If you want others to benefit from any changes please let me know.

PR #15682 (2019-10-28): Added BIQU TANGO V1 Support
### Description

These three changes offer support for the BIQU Tango V1 controller board.  The Tango board is based on the Rumba board with the exception of a few digital pin changes for fans and analog inputs.

### Benefits

Board output Fan 0 is now the extruder fan. 
Board output Fan 1 is now the FAN_PIN
Analog A, and Analog B digital pins are adjusted as well.
 
### Related Issues
Fixes #15651
Fixes #13567

Note: Without these changes :
1.) The installer would likely experience thermal runaway because the part cooling fan is swapped with the Extruder.  
2.) Ins [... cut]