MarlinFirmware/Marlin #13570: [BUG] SKR 1.3 not able to control printer through USB

spilz87, 2019-04-04:
### Description
When I connect my printer through USB to my PC, a serail port com is detected, but I can't use it to control my printer.

### Steps to Reproduce

- Connect the printer to the PC with USB cable
- Open HyperTerminal
- select right baudrate
- open port com
- send G-code command

**Expected behavior:** [What you expect to happen]
be able to control the printer through USB

**Actual behavior:** [What actually happens]
-> nothing happend ...

Same steps with an external USB-SERIAL adapter connected to RX-TX connector works well.

#### Additional Information

I can guess the issue comes from my Configuration.h file :

#define SERIAL_PORT 0
//#define SERIAL_PORT_2 -1

but I don't know how to modify it as I still want to be able to acces to the control of the printer through RX-TX connector.

is there a way to have tue same data sent from printer to USB and TX ? and be able to send data from PC to printer through USB and other data through RX ?

If yes, how to configure SERIAL_PORT and SERIAL_PORT_2 for this board ?

thanks for your help

jjansen85, 2019-04-04:
According to https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/firmware/Marlin-bugfix-2.0.x/Marlin/Configuration.h it is 
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

For me this is working perfectly with the SKR V1.3 in combination with Octoprint on a raspberry pi.

spilz87, 2019-04-04:
Thanks, it works 
But I don’t get the same data in both : 
If I send a command on RX I would like to receive echo/answer on both

p3p, 2019-04-04:
That's not how it works, it's two separate communication channels, the protocol would break down if data / ok responses were sent to both apart from specific commands that trigger automatic continuous responses.

spilz87, 2019-04-05:
Ok thanks for explanations 
But what about data sent by printer when print from SD card like echo etc. 
Is there a way it send on both or not ?
(Isn’t like spontaneous data, not a reply to a command)

closed 2019-04-05 by spilz87

jarjartee, 2019-05-29:
same issue experienced as described above

doggyfan, 2019-05-29:
Setting a serial port to '-1' on a 32-bit system like an LPC1768 board (or an STM32F) sets the serial port to use USB (baudrate is not really used, it's always fast). Setting a serial port to'0' on the same system sets it to use the (often much slower) hardware serial UART port connector on the board.

You normally have the main port (SERIAL_PORT) set to use the USB as per following, though I don't suppose it really matters which one is which ..

```cpp
#define SERIAL_PORT    -1
#define SERIAL_PORT_2 0
```