MarlinFirmware/Marlin #13571: [BUG] FAST_PWM_FAN does not increase frequency on Pin9

holgin, 2019-04-04:
### Description
I am using a custom board in my 3D printer, although for the PWM Fan I am using Pin9, so it is a default option, I guess.
The fan is very loud at this moment and I figured it is because of the low PWM frequency. Oscilloscope confirmed that I have 488Hz (50% duty cycle) on Pin9.
I tried to increase the frequency using `FAST_PWM_FAN` but it has no effect.
I did some changes in my `pins_EMS.h` file, compared to original RAMPS config file (deleting sections that I am not using, for example servo section), but IN THEORY it should not matter (unless I have deleted something important).

Of course #define FAN_SOFT_PWM is not defined (commented).
[files.zip](https://github.com/MarlinFirmware/Marlin/files/3044944/files.zip)

### Steps to Reproduce

1. configure pins_RAMPS.h to use Pin 9 for FAN_PIN
2. uncomment #define FAST_PWM_FAN in configuration.h.
3. measure the frequency with an oscilloscope

**Expected behavior:** 
Get ~8kHz PWM at Pin9

**Actual behavior:**
I'm stuck at 488Hz

#### Additional Information

Checked on both "normal" and bugfix 1.1.9 release.

dkaukov, 2019-04-16:
Same behavior with MKS GEN L. I've tried to debug it little bit. The fast pwm is actually set correctly here: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/temperature.cpp#L1127 (if I add delay there and analogWrite() I can see with oscilloscope that it set 32Khz ) But it gets reset later on somewhere. Probably something competing for TIMER2.

dkaukov, 2019-04-16:
Ok. It is actually conflicting with Tone library. Disabling "SPEAKER" restores FAST_PWM_FAN on pin 9 (at least on mega 2560)

holgin, 2019-04-16:
Thank you very much! I will try it after my print finishes.

holgin, 2019-04-17:
Tested it, the PWM frequency has increased, the fan runs much smoother now.

closed 2019-04-17 by holgin