MarlinFirmware/Marlin #13572: [FR] modify action of rotary encoder on Home screen

spilz87, 2019-04-04:
### Description

I would like to change the action of rotate rotary encoder on home screen.

### Steps to Reproduce

On home screen, when you turn the rotary encoder, the speed is changed.
I would like to be able to change something else like fan speed or flux. 

#### Additional Information

I understand it’s a specific request and everyone don’t need/want it. 

Just someone can tell me where this action is coded so I can modify it myself ?

Thanks for your help

Regards

drzejkopf, 2019-04-04:
https://github.com/MarlinFirmware/Marlin/blob/83e214478e8f945acfe75b620b193f83341225a3/Marlin/src/lcd/ultralcd.cpp#L462

It's here I believe

spilz87, 2019-04-05:
It’s exactly what I was looking for, I didn’t found it in Marlin 1
Thanks a lot

closed 2019-04-05 by spilz87