MarlinFirmware/Marlin #13576: [BUG] Speeds increased with newest TH3D marlin firmware

Silithas, 2019-04-05:
### Description

I updated the stock outdated creality ender 3 marlin firmware to the newest available TH3D firmware, but now everything with the printer has accelerated compared to the old.  Homing is now 2, if not 3 times faster than stock

### Steps to Reproduce

Update from old stock ender 3 marlin to the newest from TH3d

**Expected behavior:** [What you expect to happen]

I expected it to be updated with the newest failsafe settings in place and stock speeds

**Actual behavior:** [What actually happens]

Everything from X, Y, Z and even extruder speeds has increased.

#### Additional Information

Tried printing with usb and through octopi, but it seems to be FW settings that has changed as i'm using the same old slicing settings in cura 4.

ManuelMcLure, 2019-04-05:
This is something that needs to be taken up with the creators of TH3D - they are the ones that are providing the configuration files. You can try using vanilla Marlin with the included Ender 3 configuration file and see if that gives better results.
This is not a problem for this bug tracker.

swilkens, 2019-04-06:
The ender 3 firmware configuration of Marlin has reduced home speeds, that is probably the difference you are noticing.

This reduced homing speed was done intentionally to prevent hardware damage.

Original Issue report:

https://github.com/MarlinFirmware/Marlin/issues/12074

And the change made:

https://github.com/MarlinFirmware/Marlin/commit/06e2afe8e0c77043bbee227a59a82214c62fb1f2

swilkens, 2019-04-10:
We have since changed the homing speeds on all Creality models in our example database. The other differences you are seeing are questions for the TH3D developers.

If this has answered your question, please close this issue report :)

closed 2019-04-10 by thinkyhead