MarlinFirmware/Marlin #13578: [BUG] Homing backoff doesn't compile

doggyfan, 2019-04-05:
Hello

Just went to compile the current bugfix 2.0 branch and found a missing variable (endstop_backoff) in G28.cpp when homing backoff is enabled (HOMING_BACKOFF_MM in Configuration_adv.h).

Part from that, good work!

=======

sketch\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28(bool)':

sketch\src\gcode\calibrate\G28.cpp:410:39: error: 'endstop_backoff' was not declared in this scope

     const float backoff_x = doX ? ABS(endstop_backoff[X_AXIS]) * (X_HOME_DIR) : 0,

                                       ^

sketch\src\gcode\calibrate\G28.cpp:413:9: error: 'backoff_z' was not declared in this scope

     if (backoff_z) do_blocking_move_to_z(current_position[Z_AXIS] - backoff_z);

         ^

sketch\src\gcode\calibrate\G28.cpp:414:22: error: 'backoff_y' was not declared in this scope

     if (backoff_x || backoff_y) do_blocking_move_to_xy(current_position[X_AXIS] - backoff_x, current_position[Y_AXIS] - backoff_y);

=======

laszloh, 2019-04-05:
I can confirm, that Marlin does not compile with HOMING_BACKOFF_MM enabled. 

My error log:
>Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28(bool)':
Marlin\src\gcode\calibrate\G28.cpp:410:39: error: 'endstop_backoff' was not declared in this scope
const float backoff_x = doX ? ABS(endstop_backoff[X_AXIS]) * (X_HOME_DIR) : 0,
^

The fix is simple: https://github.com/laszloh/Marlin/commit/c2155db6d1914c293244952285c13b38469c0740

With the change, Marlin compiles, but I have not yet tested it on a live system.

marcio-ao, 2019-04-05:
I made a PR for this fix.

marcio-ao, 2019-04-05:
https://github.com/MarlinFirmware/Marlin/pull/13585

doggyfan, 2019-04-07:
Good work :)

marcio-ao, 2019-04-09:
@doggyfan: Could you close this ticket if it has been fixed?

doggyfan, 2019-04-09:
It showed as closed 3 days ago for me ?

closed 2019-04-09 by doggyfan

marcio-ao, 2019-04-09:
@doggyfan: The ticket was open, the pull request with the fix was closed :) They're both closed now. Thank you!