MarlinFirmware/Marlin #13583: [FR] include an analogue of M119 g-code to menu: "Test End Switches"

ShaggyDog18, 2019-04-05:
### Description
include an analogue of M119 g-code to menu, kind of "Test End Switches" menu item, so that it is possible to make a quick test and see the status of all end switches (at once r sequentially) at the display without any additional tool. That would be helpful to debug end switches and tune up inversion as well as direction of motors.

### Example of Menu
Test End Switches
!- Test X-end switch
!- Test Y-end switch
!- Test Z-end switch
!- Test Filament Run-out
!- Status of all end switches (M119)

7eggert, 2019-04-06:
I'd rather like the status of end switches to be displayed all at once continuously in that menu.

Also the feature to blink "?" could be extended to optionally display "!" while a single end switch per axis is pressed (most common setup?). If you have more than one end switch per axis, it could display a cycle of e.g. "Z→?→1→2→Z" as appropriate. Having a min and a max end switch, it could be "Z→?→+→-→Z". (This would not replace having a endstop status screen).

sjasonsmith, 2020-09-11:
This seems very useful to me, I'm a little surprised it doesn't already exist.

thisiskeithb, 2024-09-05:
Added in https://github.com/MarlinFirmware/Marlin/pull/25667