MarlinFirmware/Marlin #13586: [IDEX DXC_AUTO_PARK_MODE Issue ] (At a tool change the new tool goes to the previous tools last position)

dp90pusika, 2019-04-05:
### Description

During a print if a tool change command is sent, the tool change happens but the new tool goes to the last printing position of the previous tool head.
Basically the last XY coordinate is stuck, into the planner or something.

Older versions did't have this issue, the next tool was going to the next position after the tool change.

### Steps to Reproduce

Just start a print with two bodies on an IDEX machine.
One bides with T0 and one with T1.
The print starts normally, but at a tool change event, the new extruder touches the last extrudes end position on the wrong body. 

**Expected behavior:** [What you expect to happen]

IDEX is best to prevent color cross contamination at dual color prints, so after a tool change the new tool has to go to the next line in the g code, to not touch the end position of the previous head, and leave a discolored dot on the model.

**Actual behavior:** [What actually happens]

Right now the two separate bodies with different colors are contaminating each in ewer layer at the tool change coordinate.

#### Additional Information

Using [bugfix-2.0.x] downloaded at april 2.
RADDS board and Arduino due controler
Using X2 for IDEX Y2 for a second Y motor for more torque and using Z2 and dual Z endstops.
ALL stepper drivers are tmc2130 properly hooked up on SPI.

InsanityAutomation, 2019-04-08:
This has been the function in auto park since I started dealing with tool change in 1.1.8 and is currently coded this way by design. Its a "safe" action that leaves a head where its expected to be in case the next move is not a travel move. It may be possible to look ahead at the next command and determine if the next move is then go there instead, but for now its the only way to guarantee itll work regardless of gcode. Full control mode does allow you to unpark to wherever youd like though.

dp90pusika, 2019-04-22:
OK guys, this is definitely broken,

I started to use the DXC_FULL_CONTROL_MODE and it does the same thing basically.
I attach a video of how a tool change looks like and also my g code that runs.
It should not behave like this, but the last coordinate is stuck whatever I try, so the two heads are crushing in each other. 
I'll also attach my tool change script, I think there is nothing I can do to fix it with only the g-code.

Several years ago I used a different branch of Marlin, and that worked as it should be, but that branch stopped updating. It would be really nice to have back that working state.

This is the link for the branch that worked fine:
https://github.com/bobc/Marlin/tree/32-Bit-RCBugFix

### My g-code:
[modified for 3DP1.txt](https://github.com/MarlinFirmware/Marlin/files/3103195/modified.for.3DP1.txt)

### My tool change script:

`{IF OLDTOOL=0}G1 X0 F4000;
{IF OLDTOOL=1}G1 X438.15 F4000;

T[new_tool] ; switch to new tool

{IF NEWTOOL=0}G1 X0 F4000;
{IF NEWTOOL=0}G91
{IF NEWTOOL=0}G1 E10.0000 F480
{IF NEWTOOL=0}G90

{IF NEWTOOL=1}G1 X438.15 F4000;
{IF NEWTOOL=1}G91
{IF NEWTOOL=1}G1 E10.0000 F480
{IF NEWTOOL=1}G90`

### Video of the behavior:
Right extruder is T0 left is T1
https://youtu.be/g9tSdXruqoY

Pleas if you see anything I missed, anything I can do to fox this let me know.
Thank you!

InsanityAutomation, 2019-04-22:
Theres another open issue on the breakage of full control mode right now. I was able to reproduce the behavior over the weekend, but didnt have time to debug. Its on the list to work on shortly.

InsanityAutomation, 2019-04-23:
Please check if the current snapshot resolves the issue on your end. If so, please close this issue.

Thanks

dp90pusika, 2019-05-01:
Hi @InsanityAutomation 

Thank you for the fix.
The latest snapshot seems to work as expected in DXC_FULL_CONTROL_MODE.

closed 2019-05-01 by dp90pusika