MarlinFirmware/Marlin #13590: [BUG] 2.0.x Tool change / Nozzle Change compiling error

schuessm, 2019-04-06:
### Description

When I'm compiling Marlin 2.0.x with activated servo I'll got the following error. I just use marlin 2.0.x from 5th of april 2019. I want use the servo port 0 (PIN11) to switch/lift my nozzle by one 9g-servo. 

Output from compiling:

`In file included from sketch/src/module/tool_change.cpp:53:0:
sketch/src/module/tool_change.cpp: In function 'void move_nozzle_servo(uint8_t)':
sketch/src/module/tool_change.cpp:124:85: error: 'e' was not declared in this scope
       MOVE_SERVO(SWITCHING_NOZZLE_SERVO_NR, servo_angles[SWITCHING_NOZZLE_SERVO_NR][e]);
                                                                                     ^
sketch/src/module/servo.h:93:40: note: in definition of macro 'MOVE_SERVO'
 #define MOVE_SERVO(I, P) servo[I].move(P)
                                        ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.`

This is my actual configuration-file
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3050188/Config.zip)

InsanityAutomation, 2019-04-08:
Check with PR #13614 - I do not have a switching extruder to test with but it looks correct now.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle