MarlinFirmware/Marlin #13592: [BUG] PlatformIO Autobuild not working - manual PIO build works fine

ManuelMcLure, 2019-04-06:
### Description

After installing the Autobuild extension into VSCode per the instructions at [http://marlinfw.org/docs/basics/install_platformio_vscode.html](http://marlinfw.org/docs/basics/install_platformio_vscode.html), the autobuild buttons do not work. A manual PIO build does work fine. Tested with both LPC1768 (for Re-Arm/RAMPS) and megaatmega2560 (for Arduino Mega/RAMPS).

### Steps to Reproduce

1. Install VSCode, Autobuild and PlatformIO following the instructions at the URL above
2. Try to build Marlin by selecting the "B" icon in the AB menu.

**Expected behavior:**

Marlin builds - if megaatmega2560 is selected, a window will pop up asking for which processor (1280 or 2560) to use.

**Actual behavior:**

Nothing seems to happen at first glance. However, if you look at the VSCode Terminal, a new terminal (that is not the default displayed one) is created, with the following content:
```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

PS D:\Git\MarlinReArm> python buildroot/share/atom/auto_build.py build
python : The term 'python' is not recognized as the name of a cmdlet,
function, script file, or operable program. Check the spelling of the name, or
if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ python buildroot/share/atom/auto_build.py build
+ ~~~~~~
    + CategoryInfo          : ObjectNotFound: (python:String) [], CommandNotFo
   undException
    + FullyQualifiedErrorId : CommandNotFoundException

PS D:\Git\MarlinReArm>
```
It appears that VScode is trying to run the autobuild stuff in a plain PowerShell instead of one with the PlatformIO stuff on the PATH. If I type "python" into that shell I get a similar error. However, if I open a PowerShell by clicking on "PlatformIO: New Terminal" icon in the bottom bar I get a PowerShell that does understand where python is.

reloxx13, 2019-04-06:
pls try starting vsc as admin

reloxx13, 2019-04-06:
i cannot remeber if i did this for marlin or some other project.

add python to PATH
u can do this in systemcontrol -> system -> advanced systemsettings (in left sidebar) and than in the new popup advanced tab -> is a button with sth like variables at the bottom.
now a new popup will open, doubleclick PATH at the top entries (user variables) and add the path 
`C:\Users\YOUR_USERNAME\.platformio\python27\`
`C:\Users\YOUR_USERNAME\.platformio\python27\Scripts`

[image]

[image]

[image]

[image]

ManuelMcLure, 2019-04-07:
Starting as administrator didn't affect anything. I can see that adding PlatformIO's Python installation to the system PATH might help, but this can be a problem if you have a different Python install that you are using for other work. 

This worked for me until either a VSCode or a PlatformIO update broke it.

ManuelMcLure, 2019-04-07:
I feel somewhat sheepish - I didn't notice that there was a notification bell in the bottom right corner with a message about allowing the workspace to modify the terminal. Once I approved that Autobuild started working. Closing the issue.

closed 2019-04-07 by ManuelMcLure

Bob-the-Kuhn, 2019-04-08:
Please provide a screenshot of the notification bell so that it can be added to the instructions.

ManuelMcLure, 2019-04-08:
Will do as soon as I'm off work and at my machine with VSCode and PIO installed.
I was thinking that it might make sense to change the name of the new terminal windows that are opened - right now they're just "#1", "#2", etc. I would change them to something like "Autobuild 1" - I can do a PR if you think that makes sense.

ManuelMcLure, 2019-04-09:
So here's the steps I took to reproduce this. PlatformIO and Autobuild were already installed into VSCode.

- Closed all workspaces in VSCode and exited VSCode
- git cloned a new Marlin and checked out bugfix-2.0.x
- Used the PlatformIO "Open Project" option to open up the newly checked out Marlin.
- Clicked on the Autobuild icon, and then on the "B" for build.
- Nothing seemed to happen except for a new terminal labeled "#1" with the above described python   and the number 1 beside the bell icon in the bottom right of the VSCode window:
[image]
- Upon clicking the bell, I saw the following notification show up (2 images because I had to scroll down to see the whole thing):
[image]
[image]
- Clicked "Allow"
- Selected "B" from Autobuild again and got the expected "1280 or 2560 CPU" dialog box and the build succeeded.