MarlinFirmware/Marlin #13594: [BUG] Marlin 2.0 - Setting bed offsets creates unwanted behavior

paulpenney, 2019-04-06:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

swilkens, 2019-04-06:
Something seems to have gone wrong with your bug report, please update the original post with the actual description of your issue.

When you do, please also ZIP your `Configuration.h` and `Configuration_adv.h` files and attach that ZIP file to the post.

paulpenney, 2019-04-06:
> Something seems to have gone wrong with your bug report, please update the original post with the actual description of your issue.
> 
> When you do, please also ZIP your `Configuration.h` and `Configuration_adv.h` files and attach that ZIP file to the post.

Thanks, corrected.

GMagician, 2019-04-06:
Try setting MANUAL_?_HOME_POS equal to ?_MIN_POS

gloomyandy, 2019-04-06:
Please post a picture of whatever display it is you are referring to in the bug report.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@paulpenney is this still an issue?

paulpenney, 2019-09-25:
Negative. 

From: Bo Herrmannsen
Sent: September 24, 2019 2:45 PM
To: MarlinFirmware/Marlin
Cc: paulpenney; Mention
Subject: Re: [MarlinFirmware/Marlin] [BUG] Marlin 2.0 - Setting bed offsetscreates unwanted behavior (#13594)

@paulpenney is this still an issue?

boelle, 2019-09-25:
oki doke, will close then

we can always reopen

closed 2019-09-25 by boelle