MarlinFirmware/Marlin #13597: [BUG] [bugfix-2.0.x] CHAMBER_HEATER_PIN compile error

the-real-orca, 2019-04-06:
### Description

When configuring my chamber temp sensor and heater in Configuration.h, I'm getting a compiler error:
`fastio_AVR.h:68:35: error: 'DIOHEATER_CHAMBER_PIN_PIN' was not declared in this scope`

### Steps to Reproduce

Configuration.h
```Cpp
  #define TEMP_SENSOR_CHAMBER 4
  #define TEMP_CHAMBER_PIN 3 // Analog Input
  #define CHAMBER_HEATER_PIN 40  // On/off pin for enclosure heating system
```

#### Additional Information
- [bugfix-2.0.x] 3221658a7819e9f7797c5df327eef88c51a990e9
- RAMPS board
- I've disabled the chip select pins otherwise D40 would be protected. But I'm getting the same error with other pins as well.

SJ-Innovation, 2019-04-08:
Change "CHAMBER_HEATER_PIN" to "HEATER_CHAMBER_PIN"

chainreaction, 2019-04-08:
Hello,
btw, can one relocate #define *_PIN lines from configuration.h to pins_*.h files according to one's board type?
I believe that having all pins reassignments in single pins file is more straightforward. Thx

InsanityAutomation, 2019-04-08:
When a few boards have standard defined pins for the feature sure, but before that happens its beneficial to sortve advertise the pin definition name.

the-real-orca, 2019-04-12:
Thanks for the hint with the wrong pin name.

I had to fix it at some other places in the code as well. I created a bugfix commit with my changes -> see pull request above

closed 2019-04-12 by the-real-orca

chainreaction, 2019-04-15:
Ohh, great! Now I can imagine why M141 is still "unknown command". "CHAMBER_HEATER_PIN" and "HEATER_CHAMBER_PIN" are existing together somewhere within the code!

thinkyhead, 2019-04-16:
The sanity check will catch that.
```cpp
#elif defined(CHAMBER_HEATER_PIN)
  #error "CHAMBER_HEATER_PIN is now HEATER_CHAMBER_PIN. Please update your configuration and/or pins."
```