MarlinFirmware/Marlin #13600: [BUG] Compile error when enabling SPINDLE_LASER_ENABLE

s3m0, 2019-04-06:
### Description

Uncommented  SPINDLE_LASER_ENABLE
Added pins:
#define SPINDLE_DIR_PIN           P2_13   // dont Care...
#define SPINDLE_LASER_PWM_PIN     P1_23  
#define SPINDLE_LASER_ENABLE_PIN  P4_29   // SW6  Pin should have a pullup!

Got the following error:
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:43,
from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL_LPC1768/SanityCheck.h:29:6: error: #error "SPINDLE_LASER_ENABLE requires SPINDLE
_LASER_ENABLE_PIN."
#error "SPINDLE_LASER_ENABLE requires SPINDLE_LASER_ENABLE_PIN."

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

s3m0, 2019-04-06:
I am using MKS_SBASE

s3m0, 2019-04-08:
Use: #define SPINDLE_LASER_ENABLE SPINDLE_LASER_ENABLE
To enable SPINDLE_LASER_ENABLE

closed 2019-04-08 by s3m0

InsanityAutomation, 2019-04-08:
FYI, the macro expansion issue is addressed in PR #13614