MarlinFirmware/Marlin #13602: [BUG] [bugfix-2.0.x] enable AUTO_BED_LEVELING_UBL cause M502 no response

scott0122, 2019-04-07:
### Description

enable AUTO_BED_LEVELING_UBL cause M502 no response

before AUTO_BED_LEVELING_UBL enabled and send M502
[image]

after AUTO_BED_LEVELING_UBL enabled and send M502
[image]

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/3051152/conf.zip)

Ludy87, 2019-04-07:
it is written: "echo: hardcoded Default Settings Loaded". 🤔

What should your opinion still come from?

scott0122, 2019-04-07:
> it is written: "echo: hardcoded Default Settings Loaded". 🤔
> 
> What should your opinion still come from?

***before*** AUTO_BED_LEVELING_UBL ***enabled*** and send M502

swilkens, 2019-04-10:
The pictures are the same. In all your pictures, `M502` responds with `Hardcoded Default Settings Loaded`. This is correct.

What is wrong in your opinion? M502 will only load the hardcoded settings, nothing else. Are you expecting more to happen?

scott0122, 2019-04-10:
> The pictures are the same. In all your pictures, `M502` responds with `Hardcoded Default Settings Loaded`. This is correct.
> 
> What is wrong in your opinion? M502 will only load the hardcoded settings, nothing else. Are you expecting more to happen?

Sorry, I attached the wrong picture. Fixed~

closed 2019-05-18 by thinkyhead