MarlinFirmware/Marlin #13606: [1.1.x-bugfix/1.1.9/2.x-bugix] 5x TMC 2208 - Printer Reset on Long Moves

nickmitchko, 2019-04-07:
### Description

On long moves Marlin resets and returns "start" over serial connection. 

Board: MKS Gen 1.4
Drivers: 5x TMC2208 in UART
M122: Returns addresses and serial for all drivers

### Steps to Reproduce

Any long move, MKS 1.4 resets (MEGA2560)

1. Move X-carriage to 200mm
2. Home (G28 X0 Y0)
3. watch printer reset

**Expected behavior:** Printer to home and then continue operation

**Actual behavior:** printer resets and "start" is returned in the serial

#### Additional Information

Log:
```
Recv:  T:18.35 /0.00 B:20.59 /0.00 @:0 B@:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: G91
Recv: ok
Send: G28 X0 Y0
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: M122
Recv: echo:Marlin 1.1.9
Recv: 
```

* Configuration and pinouts found here: [Repo with config files](https://github.com/nickmitchko/MarlinTMC2208MKS14)
* Pictures of failure: [Click Here](https://github.com/nickmitchko/MarlinTMC2208MKS14/tree/master/Pictures)

psavva, 2019-04-08:
You've got your steps way too high.
You've configured 256 microsteps port driver on essentially a mega2560,. There is no way the board can process it in time.

You need to change your config to 16 microstepping on all axis, and leave interpolation to 256 enabled.

This way, you can still get the benefits of 256 microsteps (quite and smooth stepper movements) from the drivers and all have positional accuracy using 16 microstepping that your board can handle...

Hope this helps

nickmitchko, 2019-04-08:
@psavva Thanks! I didn't know of the limitation, although it is only an 8-bit arduino after all.

Is there a board that is known to support this configuration? 32-bit possibly?

psavva, 2019-04-08:
Why would you want to?

You would not get any better positional accuracy.
I suggest you read this: https://www.micromo.com/technical-library/stepper-motor-tutorials/microstepping-myths-and-realities

This is the list of supported 32bit controllers.
https://github.com/MarlinFirmware/Marlin/issues/7076

If your thinking of using a lot of advanced features such as linear advance, arc support, bed levelling, all the bells and whistles, then you should consider the 32 bit option.  If it's just for the microstepping, rather just interpolate on the drivers.  You can then still have your cake and eat it :)

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle