MarlinFirmware/Marlin #13607: [BUG] TMC5160 possible spi issues, error and halt on boot.

ej0rge, 2019-04-07:
### Description
Printer is Tronxy X5S retrofitted with 24v supply and RADDS running 2.0-bugfixes pulled from git earlier today. 

Have been using watterott tmc2130 w/ hardware spi implemented by soldering thin wires to the back of the board. Has been working that way for several weeks. 

Received my TMC5160 silentstepsticks (watterott by way of digi-key), soldered on the headers, replaced the 2130 boards on my radds with the new ones, pulled the latest changes from git, changed the X and Y driver type from 2130 to 5160, compiled with arduino 1.9.0 beta build 102 and flashed. 

A second or two after booting, marlin halts. 

Console reveals some oddness in the tmc debug data and reports coil shorted and overheat on the X driver, regardless of which of my two 5160 drivers are plugged into the X position or if there is a motor cable connected to the X driver. 

Revert to tmc2130 and the printer works and prints normally. 

Testing X connection... OK
Testing Y connection... OK
00:00: X driver overtemperature warning! (1200mA)
00:00: Y driver overtemperature warning! (1200mA)
X driver error detected: 0xF7F00000
overtemperature
coil short circuit
X Y
Enabled false false
Set current 1200 1200
RMS current 1170 1170
MAX current 1650 1650
Run current 17/31 17/31
Hold current 8/31 8/31
Global scaler 256/256 256/256
CS actual 31/31 15/31
PWM scale 0 0
vsense
stealthChop false false
msteps 0 256
tstep 268435456 268435456
pwm
threshold 0 0
[mm/s] - -
OT prewarn false true
OT prewarn has
been triggered true true
off time 0 0
blank time 54 54
hysteresis
-end -3 -3
-start 1 1
Stallguard thrs 0 0
DRVSTATUS X Y
stallguard X X
sg_result 0 0
fsactive X X
stst
olb X
ola X
s2gb X
s2ga X X
otpw X X
ot X X
Driver registers:
X 0x7F:BF:E0:00
Y 0x0F:DF:E0:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

### Steps to Reproduce

Attempt to use tmc5160 silentstepstick on RADDS/Due

**Expected behavior:** [What you expect to happen]

Should work?

**Actual behavior:** [What actually happens]

Doesn't. 

#### Additional Information

[X5S.zip](https://github.com/MarlinFirmware/Marlin/files/3051253/X5S.zip)

gloomyandy, 2019-04-07:
If you have STOP_ON_ERROR (and you probably do) then I suspect the overtemp and short errors will be causing the kill called. So npw you need to to figure out why you are getting those errors! 

Some other random thoughts... 
How are you powering the board? I suspect that like some of the other TMC devices you must make sure that that VMOT is applied before the 5v/3V supply is that the case with your setup? 
Do you have the CLK pin pulled to GND (not sure if you have to do that on the 2130s or not)?

kAdonis, 2019-04-07:
You definitly need to pull CLK to Gnd https://github.com/MarlinFirmware/Marlin/issues/13544#issuecomment-478625383

wesk-ms, 2019-04-07:
Since you're getting an error saying the coil is shorted, it sounds like the 5160 boards are oriented incorrectly.  The driver chip goes down, just like the 2130s.  That means the 4 mosfets are on top...all when looking at the boards when they are installed on the RADDS board.

Also, the NC pin on the 2130 is the CLK pin on the 5160.  This needs to be grounded, as was stated above.

ej0rge, 2019-04-07:
> You definitly need to pull CLK to Gnd [#13544 (comment)](https://github.com/MarlinFirmware/Marlin/issues/13544#issuecomment-478625383)
> https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC5160_v13.pdf
> you have a wrong sense resistor value
> 
> ```
>     #define X_RSENSE     0.11
>     #define Y_RSENSE     0.11
> ```
> the Watterott drivers have 0.075Ω sense resistors
> Your Configuration_adv.h is from the an older commit

Thanks for the catch wrt rsense. I hadn't considered that. 

And yes, my configuration has limped along for a few months worth of git updates, because i didn't want to re-create it every time i built. 

I had read through that thread where it was suggested to ground the clk pin but it was not clear to me from that thread that this was necessary. 

wesk-ms my boards are not oriented incorrectly. this is not my first rodeo. 

I will ground the clk pin and fix the rsense setting and see how they do.

akurz42, 2019-04-09:
Also make sure you set the right driver type in Configuration.h like
#define X_DRIVER_TYPE  TMC5160

I've left them on 2130 because I didn't realize they have their own type already. Without that I had the exact same issues as you. I'm also using a RADDS.

Rsense and CLK to GND are a must, too, as already mentioned.

ej0rge, 2019-04-13:
Wiring the CLK pin to ground and correcting the rsense setting seems to have resolved my issues. Though, i did both of them at once, so, *shrug*. 

I did have to reverse my motor cables to get it moving in the right direction.

closed 2019-04-13 by ej0rge