MarlinFirmware/Marlin #13608: [BUG] Marlin 2.0 - MKS Gen L and MKS Touch Screen 3.2

Stady234, 2019-04-07:
### Description

Marlin 2.0 - MKS Gen L and MKS Touch Screen 3.2 - Unable to flash the update while screen is connected.

### Steps to Reproduce

1. Connect official MKS touch screen to MKS Gen L board
2. Attempt to flash
3. See Failure

**Expected behavior:** Flash with screen connected

**Actual behavior:** Fails when connected.

#### Additional Information
Below is with screen connected ::
```
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Temp\arduino_build_854017/Marlin.ino.hex"
avrdude: writing flash (146886 bytes):

Writing | #avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
```
Below is screen Disconnected ::

```
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Temp\arduino_build_854017/Marlin.ino.hex"
avrdude: writing flash (146886 bytes):

Writing | ################################################## | 100% 21.13s

avrdude: 146886 bytes of flash written
avrdude: verifying flash memory against C:\Temp\arduino_build_854017/Marlin.ino.hex:
avrdude: load data flash data from input file C:\Temp\arduino_build_854017/Marlin.ino.hex:
avrdude: input file C:\Temp\arduino_build_854017/Marlin.ino.hex contains 146886 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 15.95s

avrdude: verifying ...
avrdude: 146886 bytes of flash verified

avrdude done.  Thank you.
```

doggyfan, 2019-04-07:
That's probably because the USB<>serial chip on the MKS board shares the same hardware serial pins as the touch screen (RxD and TxD pins).

You need to unplug the touch screen to allow the USB<>serial chip to use the RxD/TxD pins properly. It's not a Marlin firmware problem.

Ludy87, 2019-04-07:
@doggyfan, You've hit the nail on the head.

Correct answer!

doggyfan, 2019-04-07:
You could if you wanted move the RxD/TxD pins for the TFT over to UART2 or UART3 on the other connectors on the MKS board if your not already using the other UART pins. Then just set that other UART port as the 2nd serial port in Marlin.

or

Add a little chip between the TFT serial pins and the MKS board to automatically disconnect the TFT pins when you plug the MKS USB into your PC/tablet or whatever.

You wouldn't have to unplug the TFT screen to do firmware updates then.

Stady234, 2019-04-08:
Thank you for the info... Is there already formatted guidance on configuring the UART on the mks?  I have never hear of this, and have no idea what to do...
The MKS team only keeps saying that "they are consulting with R&D team" to just about every ticket...

boelle, 2019-07-21:
would configs be usefull here?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

odaruc, 2020-06-05:
It has been a year but in case anyone searches, I wanted to cross-reference a detailed solution to the original question mentioned here by using the y-min/max pins as RX3/TX3 for the TFT screen, and discussion of a separate but related issue that has been fixed in marlin bugfix 2.0.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/723