MarlinFirmware/Marlin #13610: [BUG] [Bugfix-2.0.x] Wrong Retraction Length in ADVANCE_PAUSE_FEATURE when E in mm3 enabled.

soimy, 2019-04-07:
### Description

I think one of the firmware updates mysteriously enabled Volumetric extruding(E in mm3) and my printer suffered from extreme under-extrusion. After a re-calibration the E step/mm amount almost tripled. A side effect: filament change in LCD menu is unusable, retraction length and load length is way longer than setting in `Configuration_adv.h` or EEPROM filament setting, which is 700/600

### Steps to Reproduce

1. Enable `#define ADVANCED_PAUSE_FEATURE` in `Configuration_adv.h`
1. Enable LCD menu -> Configuration -> Advance settings -> Filament -> E in mm3
2. Re-calibrate E step/mm
3. LCD menu Change filament

**Expected behavior:** Filament retract distance match in `#define FILAMENT_CHANGE_UNLOAD_LENGTH` or EEPROM `Advanced Settings -> Filament -> Unload mm

**Actual behavior:** Retraction length is about Re-calibrated E step/mm divided by old E step/mm

psavva, 2019-04-08:
When you uploaded the new configuration, did you test the EEPROM?

M502 Reset current settings to defaults, as set in Configurations.h

Follow with M500 to reset the EEPROM too.

soimy, 2019-04-08:
@psavva Yes I did an EEPROM reset, and also a backup/restore using octoprint EEPROM marlin plugin.
But the point is when volumetric extruding enabled, there might be a bug in ADVANCED_PAUSE_FEATURE not take that into account.
I'll upload my `Configuration.h` & `Configuration_adv.h` tonight

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@soimy is this still an issue?

closed 2019-10-12 by boelle