MarlinFirmware/Marlin #13611: BL Touch no triggering

clever83, 2019-04-07:
I've been working on getting my BLtouch to work and for some reason I can't seem to get it to trigger.

I've googled for hours and tried a done of things but nothing has worked so far. My testing results so far:
all probe commands work, Deploy, Retract, Self test, Clear alarm.

M119 shows open.
Removing the cable from either the board or the bltouch and running M119 shows triggered.
I enabled M43 in configuration_adv.h and tried it M43 S
When it gets to trigger probe. It doesn't detect the trigger.

M43 W lists then when first entered
Recv: PIN:  11   Port: B5        SERVO0_PIN
but does not show anything when pen in pushed in.

Anything to try would be appreciated at this point.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3051641/Marlin.zip)

psavva, 2019-04-07:
I had the same issue. I had to open my BLTouch and remount the pin and magnet.

Did you by any chance open/unscrew/ pull the pin/ otherwise ta tamper with the BLTouch?

clever83, 2019-04-07:
I'm sure I did when I first got it. Hard not to play with a new toy.

psavva, 2019-04-07:
In that case you'll probably need to raise or lower the grub screw until you find a sweet spot where it triggers.  I had one of the very first BLTouch from the Kickstarter campaign which is a metal pin.  I had to redo the shrink wrap and experienced the problem you're describing.  After readjusting the grub screw several times with testing the function in between turns, I found a good range where it properly triggers consistently.

clever83, 2019-04-07:
Well. a quick attempt didn't seem work. Went from all the way in to a few turn out from where it started. It never showed triggered. How sensitive is it? and do I need to do it between test runs or can I try to turn and press while it's testing? The only thing I did was mess with the pen before it was installed

Roxy-3D, 2019-04-07:
> M119 shows open.

Yes.   The BL-Touch puts out a pulse.    Unless you have an oscilloscope, you wont be able to see the pulse.   It doesn't act like a switch and present either an 'Open' or 'Closed'.   It puts out a pulse when it gets touched.

kb79000, 2019-04-07:
It's a bl touch V3 ? it's normally it's buggy with marlin

clever83, 2019-04-07:
It's a 2.2

FanDjango, 2019-04-07:
Deploy it. It should go dark, no red light. Trigger it with your finger. If it lights up red and pulls up the pin, it has triggered and it has output the pulse. Doesn't pull up the pin? Problem lies in the probe. Does pull up the pin and lights up red? Problem lies in the white/black cable to the board, how it is connected, the firmware, the configuration etc. but most probably not the probe.

kb79000, 2019-04-07:
If you have bltouch v3 and ender 3 test this version it's OK
https://drive.google.com/file/d/1SPHg1Yins9Vp5xHHS9_Q_ORmkQsVKJAB/view?usp=drivesdk

clever83, 2019-04-07:
I can deploy and it will turn blue when pressed up, so that seems to work as it should. If that's the case your saying it's config and not probe?

FanDjango, 2019-04-07:
Just a light touch upwards and probe itself pulls up the pin - then it's working.

I see in your config

`#define BLTOUCH_DELAY 100`

I thought the default (and recommendation) was 350? This could be an issue...

And your config says

`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

so you should connect the white wire to pin 18, I would think, instead of an endstop switch you had before.

clever83, 2019-04-07:
They pdf they have on the website shows  #define BLTOUCH_DELAY 100
I have my black/white connected to the z stop on the marlin board I'm assuming thats pin 18, I don't have a pin layout, just the few guides I could find online

clever83, 2019-04-07:
Alright. So I feel dumb if this was all it was, but after pulling the cables and rechecking all the connections, It started working. Thank you everyone for your help. Now just to fix the extruder clicking.

closed 2019-04-07 by clever83