MarlinFirmware/Marlin #13615: PRINTER HALTED Please Reset

takno91, 2019-04-08:
when used lcd 128 64 error show
PRINTER HALTED Please Reset
but when used usb cable to print work fine  
i replace lcd but The problem is still there
i used 16 mega memory card and his work another printer

psavva, 2019-04-08:
Please provide the config files as per the incident template instructions.

When you say you're printing via the LCD, do you mean the SD Card?

Can you attach the gcode file your trying to print?

If you connect the usb cable and start Pronterface or Repetier Host, you will get some logs, provide that too.

thinkyhead, 2019-04-11:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-04-11 by thinkyhead