MarlinFirmware/Marlin #13617: [BUG] LPC1769 can't use ReprapDiscount GLCD

FLYmaker, 2019-04-08:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
If I enable this line of code, he will report an error. Smoothie can use it

Why can't Marlin

ManuelMcLure, 2019-04-08:
What exact board? There are several LPC1769 boards out there with different connections. What exact error? I’m using REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER just fine on my LPC1768-based Re-ARM.

FLYmaker, 2019-04-08:
> 什么确切的板？有几个LPC1769板有不同的连接。什么确切的错误？我在基于LPC1768的Re-ARM上使用REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER就好了。
Self-made, based on smoothieboard 5x

psavva, 2019-04-08:
Then I suggest checking your schematics, ensure pins are defined correctly, SPI interface on an oscilloscope, etc.

Please check https://github.com/MarlinFirmware/Marlin/issues/7076 for current supported boards.

p3p, 2019-04-08:
> Why can't Marlin

It can, and does. 

Don't ignore issue templates they are there for a reason.

closed 2019-04-08 by p3p