MarlinFirmware/Marlin #13618: MKS gen 1.4; "Smart Controller LCD 12864"; Backlight but without any image/picture/screen

TheRealVira, 2019-04-08:
To begin with, I am using the latest released marlin firmware version 1.1.9 ~ which has the option for full graphics discount reprap smart controller LCDs ~ which I have uncommented as intended inside the configuration file. The MKS gen 1.4 board has been flashed by using the latest Arduino IDE (also added u8glib inside the IDE via their built in library manager). Thus, everything compiled and uploaded successfully.

Apart, I needed to fix my display since it seemed like it got regulated by a weird resistor. It made my board force to run on 3V though it actually needs 5V to run.. so soldering out said resistor fixed my "contrast issue".

Now back to my problem - backlight *is* working and I do see the darker squares indicating the screen, but nothing else pops up. Lastly, interestingly, the selector button buzzer does in fact work and beeps on presses.

robbycandra, 2019-04-08:
I think this is hardware problem. Not related to marlin.

ManuelMcLure, 2019-04-08:
MKS is notorious for putting the shrouds around the display connector on backwards. Try shaving the "key" off the connector on the cable and plugging them in the other way around.

TheRealVira, 2019-04-08:
I have noticed that I need to comment out CR10 default lcd - marlin does not support multiple LCD drivers.

closed 2019-04-08 by TheRealVira