MarlinFirmware/Marlin #13619: U8Glib v 1.19.1 Causes Errors must revert libary back to 1.18.0

Xeliaa, 2019-04-08:
Error of newer u8glib down below which is v1.19.1, have to revert to 1.18.0 to be able to compile,
github for olivers u8glib says its not going to be maintained and to use U8g2lib now?
=======================
Build options changed, rebuilding all
C:\Users\saeba\OneDrive\Documents\Arduino\libraries\U8glib\src\clib\u8g_rect.c: In function 'u8g_DrawBox.constprop':

C:\Users\saeba\OneDrive\Documents\Arduino\libraries\U8glib\src\clib\u8g_rect.c:126:1: internal compiler error: Segmentation fault

}

^

Please submit a full bug report,

with preprocessed source if appropriate.

See http://gcc.gnu.org/bugs.html for instructions.

lto-wrapper.exe: fatal error: C:\Users\saeba\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/saeba/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

ManuelMcLure, 2019-04-08:
A segmentation fault in the compiler is a compiler bug - there's nothing the Marlin developers can do to fix this. You need to bring this up with the developers of the compiler.

psavva, 2019-04-08:
Try compiling with platformio or Arduino.  
Suggest you close this in issue in this Marlin repo, and open one as per http://gcc.gnu.org/bugs.html

closed 2019-04-09 by Xeliaa