MarlinFirmware/Marlin #13623: [BUG] Garbled display when using LIGHTWEIGHT_UI and steppers deactivate while in Z-Offset screen

marcio-ao, 2019-04-08:
### Description

Garbled display when using LIGHTWEIGHT_UI and steppers deactivate while in Z-Offset screen

### Analysis

Since the LIGHTWEIGHT_UI  uses the text buffer rather than the graphics buffer, it needs to be notified when Marlin enters and exits the status screen so that it can properly clean up after itself.

This notification is handled by the `MarlinUI::update()` method which monitors `currentScreen` and triggers a clear of the screen when transitioning in and out of `status_screen`.

However, on stepper timeout, the status screen gets painted directly outside of `MarlinUI::update()`:

```
        #if HAS_LCD_MENU
          ui.status_screen();
          #if ENABLED(AUTO_BED_LEVELING_UBL)
            if (ubl.lcd_map_control) {
              ubl.lcd_map_control = false;
              ui.defer_status_screen(false);
            }
          #endif
        #endif
```

This code was added in commit eb4bf93ea4 and it is unclear to me what the intent was. There are two possible solutions, depending on what was intended.

### Solution #1:

Replace `ui.status_screen()` with `ui.return_to_status()'. On stepper timeout, if the user was in the menu, they will be kicked back into the status screen.

### Solution #2

Only call `ui.status_screen()` if the user is already in the status screen:

```
        #if HAS_LCD_MENU
          if(ui.on_status_screen())
             ui.status_screen();
          #if ENABLED(AUTO_BED_LEVELING_UBL)
            if (ubl.lcd_map_control) {
              ubl.lcd_map_control = false;
              ui.defer_status_screen(false);
            }
          #endif
        #endif
```

### Solution #3

Remove the call to `ui.status_screen()`. It's unclear what it is accomplishing. If the intent was to draw the "?" for the position, I suspect this will get done automatically on the next LCD update. Without the redraw, the code would look like this:

```
        #if HAS_LCD_MENU && ENABLED(AUTO_BED_LEVELING_UBL)
            if (ubl.lcd_map_control) {
              ubl.lcd_map_control = false;
              ui.defer_status_screen(false);
            }
        #endif
```

@InsanityAutomation: The commit was yours. Can you shed any light on why you are repainting the status screen and whether any of the above solutions would be acceptable?

### Steps to Reproduce

1. Enable LIGHTWEIGHT_UI
2. Go into Z-offset screen
3. Wait a really long time for the steppers to inactivate
4. Notice that the screen is now garbled. It will remain garbled until you click to enter a menu, click again to exit the menu.

Another place where this happens is that if the user is playing one of the new games, the display will become garbled if the steppers timeout.

**Expected behavior:** On timeout, it should clear the graphics buffer prior to painting the text for the status screen.

**Actual behavior:** On timeout, the status screen is painted, which overlays the text on top of the graphics elements from the Z-offset screen.

InsanityAutomation, 2019-04-08:
Solution #1 was the intended action here. Probably grabbed the wrong function here.

marcio-ao, 2019-04-08:
I'm about to leave for the day, but just curious as to why you want to kick the user back to the status screen when the steppers timeout? What if they are about to get a high score in Space Invaders? :)

InsanityAutomation, 2019-04-08:
From the babystepping screen, we dont want them adjusting z offset endlessly to a value that then rams through the bed on the next move when theyre active again :)

InsanityAutomation, 2019-04-09:
My initial thought was to defer stepper timeout like advanced pause does, but havnt had time to get back to that.

marcio-ao, 2019-04-09:
I'll see if I can whip something together.

marcio-ao, 2019-04-09:
See my new PR https://github.com/MarlinFirmware/Marlin/pull/13630

marcio-ao, 2019-04-09:
@InsanityAutomation: So whenever the stepper de-energizes, the accuracy loss is far greater than any tuning the user may have done through babystepping? If that is the case, it doesn't matter if the user is returned to the main screen on stepper timeout, because their fine-tuning is borked from that point forward anyway.

I think it would be more correct to say BABYSTEP_ALWAYS_AVAILABLE is fundamentally incompatible DISABLE_INACTIVE_Z ... why not simply have the sanity check prevent these features from being used together?

marcio-ao, 2019-04-09:
> however with either of our firmware builds this means that when steppers are de-energized we may lose z position drifting slightly so it can no longer be relied on.

For what it is worth, we no longer have DISABLE_INACTIVE_Z enabled on our Mini 2 or TAZ Pro.

InsanityAutomation, 2019-04-09:
Short term incompatible avoids an issue there, but at some point ill do a proper defer like is done for parking. Personally I wouldnt want to leave a stepper humming all the time. A burned out driver has been the only major failure on one of my Taz6 machines after all.

marcio-ao, 2019-04-09:
@InsanityAutomation: Well, your original implementation wasn't doing what you thought it was doing anyway, so I'll let this PR stand. If you wish, you can replace `ui.status_screen()` with `ui.return_to_status()` on your customized FW to get the behavior you intended, but I don't think that ought to happen in the official release as it is incorrect to return the user to the status screen _from any screen_ when the steppers de-energize.

InsanityAutomation, 2019-04-09:
Yup, hence why I agreed with it above for now and ill get back to doing it completely some other time. There are likely other screens we would want them to return to status from though. The UBL mesh edit screens are a prime example.

marcio-ao, 2019-04-12:
Closing this since it has been fixed.

closed 2019-04-12 by marcio-ao