MarlinFirmware/Marlin #13625: [Maintenance] Reduce repository size

p3p, 2019-04-09:
Marlins git repository is getting a little on the beefy side, this is an example of how that could be mitigated if @thinkyhead thinks it is worth it. (I'm not a git expert by any means though)

```console
p3p@Zeus ~/workspace $ git clone git@github.com:MarlinFirmware/Marlin.git
Cloning into 'Marlin'...
remote: Enumerating objects: 124, done.
remote: Counting objects: 100% (124/124), done.
remote: Compressing objects: 100% (97/97), done.
remote: Total 171236 (delta 60), reused 38 (delta 25), pack-reused 171112
Receiving objects: 100% (171236/171236), 100.22 MiB | 4.10 MiB/s, done.
Resolving deltas: 100% (109847/109847), done.
Checking connectivity... done.
p3p@Zeus ~/workspace $ cd Marlin
p3p@Zeus ~/workspace/Marlin $ git count-objects -vH
count: 0
size: 0 bytes
in-pack: 171236
packs: 1
size-pack: 104.80 MiB
prune-packable: 0
garbage: 0
size-garbage: 0 bytes
```

This can be mitigated by using gits garbage collection tools:
```console
p3p@Zeus ~/workspace/Marlin $ git reflog expire --expire=now --all && git gc --prune=now --aggressive
Counting objects: 171236, done.
Delta compression using up to 8 threads.
Compressing objects: 100% (158543/158543), done.
Writing objects: 100% (171236/171236), done.
Total 171236 (delta 118068), reused 46777 (delta 0)
p3p@Zeus ~/workspace/Marlin $ git count-objects -vH
count: 0
size: 0 bytes
in-pack: 171236
packs: 1
size-pack: 48.24 MiB
prune-packable: 0
garbage: 0
size-garbage: 0 bytes
```
This reduces the local repository by about 50% but we really need to be cleaning the remote so lets work on a mirror repo that when pushed will overwrite all references.

```console
p3p@Zeus ~/workspace $ git clone --mirror git@github.com:MarlinFirmware/Marlin.git
Cloning into bare repository 'Marlin.git'...
remote: Enumerating objects: 304, done.
remote: Counting objects: 100% (304/304), done.
remote: Compressing objects: 100% (251/251), done.
remote: Total 276718 (delta 105), reused 53 (delta 34), pack-reused 276414
Receiving objects: 100% (276718/276718), 163.63 MiB | 4.06 MiB/s, done.
Resolving deltas: 100% (176812/176812), done.
Checking connectivity... done.
p3p@Zeus ~/workspace/Marlin.git $ git count-objects -vH
count: 0
size: 0 bytes
in-pack: 276718
packs: 1
size-pack: 171.02 MiB
prune-packable: 0
garbage: 0
size-garbage: 0 bytes

```

So the starting point is 171MiB for a repository mirror, 

```console
p3p@Zeus ~/workspace/Marlin.git $ git reflog expire --expire=now --all && git gc --prune=now --aggressive
[... 142 lines cut ...]

In total, 51432 object ids were changed. Full details are logged here:

	/home/p3p/workspace/Marlin.git/..bfg-report/2019-04-09/01-36-27

BFG run is complete! When ready, run: git reflog expire --expire=now --all && git gc --prune=now --aggressive

p3p@Zeus ~/workspace/Marlin.git $ git reflog expire --expire=now --all && git gc --prune=now --aggressive
Counting objects: 276770, done.
Delta compression using up to 8 threads.
Compressing objects: 100% (259329/259329), done.
Writing objects: 100% (276770/276770), done.
Total 276770 (delta 192241), reused 54991 (delta 0)
p3p@Zeus ~/workspace/Marlin.git $ git count-objects -vH
count: 0
size: 0 bytes
in-pack: 276770
packs: 1
size-pack: 67.46 MiB
prune-packable: 0
garbage: 0
size-garbage: 0 bytes
```

So that gets the repository mirror down to 67MiB down from 171MiB, and a normal remote clone down to 43MiB. There are a few more binaries that could be removed but are protected by the old version branches (as seen in the bfg output) so this could be reduced further. (if you don't protect the 1.0.x branch  60MiB mirror and 35MiB remote clone)

Roxy-3D, 2019-04-09:
It is real good to know this.   It gives us a number of options.    

But my starting position is we keep everything until we are forced to prune the tree.    (That doesn't mean we shouldn't fully discuss and debate this.   My vote can be changed.)

oysteinkrog, 2019-04-10:
Rewriting history will change all commit hashes, invalidating an awful lot of links that are floating around.

p3p, 2019-04-10:
> rewriting git history is bad

I could have perhaps been a little more specific with the details ^^, This was just an example of the tools available to reduce the download size mentioning the more extreme options at the end, the majority of the size reduction and general cleanup is achieved without having to resort to removing the 'large' unused blobs from history, just using `git gc`.

thinkyhead, 2019-04-10:
According to my understanding, GitHub automatically does pruning and garbage collection on the repositories on their servers. Thus you cannot reach orphan commits. They are gone.

p3p, 2019-04-10:
@thinkyhead, this isn't (can't be) true, as you can see from my post above running a normal `git gc` on a mirror of the Marlin repository halved it's size, there is a lot of redundant data.

thinkyhead, 2019-04-10:
There seems to be a way to add a GC button to settings. But I can't see how.
https://github.com/gitbucket/gitbucket/issues/1212

p3p, 2019-04-10:
As I said at the start of this I'm no git / github expert, I'm going to assume that github does not allow a mirror clone to be garbage collected then pushed back up (even though that doesn't change history) so I'l close this and hope that at some point githubs automated gc triggers .. even the git documentation recommends a `git gc --aggressive` every few hundred commits .. Marlin goes through that in a couple of weeks.

closed 2019-04-10 by p3p

thinkyhead, 2019-04-11:
BFG looks great. If only it could update commit IDs on the GitHub meta-data so all the old PR commit references would still work…. then we could use it on the main repo. I've been searching GitHub and so far haven't found any notes on how often they clean repos and how deeply.