MarlinFirmware/Marlin #13631: [FR, RFC] No "click to continue" on LED-equipped printers after SD-Print

7eggert, 2019-04-09:
After printing from SD, a printer with LED will wait for the user to click a button. (This does not happen after USB prints) Also I'm stupid enough to actually do this and thus the light goes out (which I needed not to).

While I just patched it out for me, I'm reluctant to just rip it out for everybody. I'd rather propose to
1)  load the default LED colors
1) 1) without waiting // no green lights; delay == 0
1) 2) after a delay without clicking // how to do that?
1) 3) with clicking // I don't like to extra-click

2) load a different user-defined LED color (which might be black aka. off aka. current behavior)
2) 1) dito ...

Which of these should be implemented? I'd favor having a green light and then to switch to the startup lights after a delay or on the first press of any key.

sjasonsmith, 2020-11-10:
@7eggert is this something that still concerns you? Is it even still an issue in current Marlin?

7eggert, 2020-11-10:
I patched it in my local repository and did not upgrade since then.

I hacked that part of the code by this↓ patch. I don't claim to understand all the full implications since I didn't care back then. When upgrading, I'll definitely want to not be forced to press a button just because my printer has lights.

```
--- a/Marlin/Marlin_main.cpp
+++ b/Marlin/Marlin_main.cpp
@@ -1194,7 +1194,7 @@ inline void get_serial_commands() {
             #if ENABLED(PRINTER_EVENT_LEDS)
               LCD_MESSAGEPGM(MSG_INFO_COMPLETED_PRINTS);
               leds.set_green();
-              #if HAS_RESUME_CONTINUE
+/*              #if HAS_RESUME_CONTINUE
                 lights_off_after_print = true;
                 enqueue_and_echo_commands_P(PSTR("M0 S"
                   #if ENABLED(NEWPANEL)
@@ -1207,7 +1207,7 @@ inline void get_serial_commands() {
                 safe_delay(2000);
                 leds.set_off();
               #endif
-            #endif // PRINTER_EVENT_LEDS
+*/            #endif // PRINTER_EVENT_LEDS
           }
         }
         else if (n == -1) {
```