MarlinFirmware/Marlin #13635: [FR] bonjour et t'il possible d'avoirs des sous numero de bluid de marlin 2.0 pour voir les nouvelle mise a jour ex marlin bugfix 2.0 rc0104 etc rc0105 ? merci

corv45, 2019-04-09:


7eggert, 2019-04-09:
If you use git to download Marlin, you can have a hexadecimal ID of the changes. You can also see a tree of the changes in the gitk program.

E.g. for today, there would already be 9 changes (click "commits" to see them and select the bugfix branch) - each change might be considered a version upgrade. Also when working with git, several people may work individually - the changes aren't ordered. Therefore version numbers aren't useful. ¢¢

corv45, 2019-04-09:
oki for commits I did not know. and for you can have a hexadecimal identifier I can not find it?

swilkens, 2019-04-10:
The list of commits shows you what was changed in time. There, you can also see the full code base at that point in time, so you can download `Marlin-2.0.x-5018fda` for example.

https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x

[image]

This is more of a git question than a Marlin question :)

InsanityAutomation, 2019-04-10:
It can be done with a git hook however every single person actually merging anything in will need to have that set up in their environment. It's been discussed before, but as far as I know there is no way to have it done automatically if you merge from the web interface, or if it's merged by a user without their environment configured for it. That just makes it very difficult to actually maintain correctly at this scale.

7eggert, 2019-04-10:
The ID is a checksum for the commit. This checksum is the ID for the code at that point in time.

There are branch names, too, like bugfix-2.0.x, they are a pointer to that checksum, and they get updated on each commit.

Also you can have tags doing about the same, except they are not updated automatically. Github will show them as releases (as far as I instant-googled).

closed 2019-04-10 by thinkyhead