MarlinFirmware/Marlin #13636: [BUG] SKR 1.3, neopixel and Marlin 2.0

DimchikUA, 2019-04-09:
### Description
For some time I use MKS gen1.3 (atmega 2560) board and marlin 1.1.9 firmware in my 3D printer.
In this configuration, the bltouch and neopixel strip are configured and work well.
Recently, I decided to switch to the 32 bit platform. I chose the [BIGTREETECH-SKR-V1.3](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-PIN.pdf) board.
This board has only one servo plug and there is no special place where you can connect additional servo or neopixel strip. Nevertheless, I have a lot of empty connectors and assumed that I could add some pin for a neopixel. While I do not really understand what to do.
 I assume that I can add the necessary pins to the file pins_BIGTREE_SKR_V1.3.h in the servo section, it looks like in the file pins_MKS_SBASE.h  and it should work.
```
// Servo pin
//
#define SERVO0_PIN         P1_23   // J8-3 (low jitter)
#define SERVO1_PIN         P2_12   // J8-4
#define SERVO2_PIN         P2_11   // J8-5
#define SERVO3_PIN         P4_28   // J8-6
```
### But all this is not a bug. Bug further.

### Steps to Reproduce
my configuration

**platformio.ini**
`env_default = LPC1768`

**Marlin\Configuration.h**
...
`#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3`
...
`#define BLTOUCH`
```// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN    4        // LED driving pin on motherboard 4 => D4 (EXP2-5 on Printrboard) / 30 => PC7 (EXP3-13 on Rumba)
  #define NEOPIXEL_PIXELS 38       // Number of LEDs in the strip
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 200  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
#endif
```

**Expected behavior:** 
I expect that with this configuration there should be a compilation error, because the SKR-1.3 board does not have a special pin for neopixel

**Actual behavior:** 
In fact, the firmware is compiled without any errors, it is not clear which pin is assigned to the neopixel strip.
I guess this is a bug.

#### Additional Information
If you comment out #define NEOPIXEL_LED, the firmware still compiles normally.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3060643/Configuration.zip)

p3p, 2019-04-09:
While pin `4` is not a correct pin number for LPC based boards that need to use `PX_XX` style pin identifiers, the integer value of 4 is still a valid pin index, the PX_XX style just hides the real value. `4` is probably the same as P0_04 as long as it has no extra features available.
We can't catch all configuration errors, sometimes we just have to trust the user to change the pin id to one that makes sense on the board they are using.

As a side note although you may have lots of free pins, makes sure they do not have any other things connected to them on the board (pullup resistors, filter circuits, etc)

closed 2019-04-09 by p3p

SISLANGER, 2019-04-27:
@DimchikUA did you get this sorted, I need to find a way for neopixel also

DimchikUA, 2019-05-01:
I used pin P1_27, he is not busy with me.  But neopixel does not work from 3.3v.  need a solution, such as https://images.app.goo.gl/RqvVrfQXfY9x1Q176 or https://images.app.goo.gl/kg6uTDYtzqn5PR1a8

SISLANGER, 2019-05-01:
Good to know

doggyfan, 2019-05-01:
P1_27 is normally the Y- (Y limit switch) on the SKR 1.3 board, so don't forget to not use Y- limit switch connector for your Y limit switch. It also has a file network on it before the pin, an RC filter (0.1uF and I can't remember the resistor value)..

DimchikUA, 2019-05-01:
I use  y+ limit switch and y- is free. That's what I used.

jmdearras, 2019-05-01:
Looking at the neopixel library, it does not appear it supports the LPC1768.

jmdearras, 2019-05-01:
> I used pin P1_27, he is not busy with me. But neopixel does not work from 3.3v. need a solution, such as https://images.app.goo.gl/RqvVrfQXfY9x1Q176 or https://images.app.goo.gl/kg6uTDYtzqn5PR1a8

does it work with a level shifter?

p3p, 2019-05-01:
@DimchikUA 
> But neopixel does not work from 3.3v. need a solution

3.3v is above the state transition for 5.0 logic so it should work in most conditions, I know I run my neopixels on 3.3v logic level pins.

@jmdearras 
> it does not appear it supports the LPC1768.

my fork that is used in LPC176x builds does

jmdearras, 2019-05-01:
> @DimchikUA
>
>
> 3.3v is above the state transition for 5.0 logic so it should work in most conditions, I know I run my neopixels on 3.3v logic level pins.
>
> @jmdearras
>
>
> my fork that is used in LPC176x [...]

you have it working on the LPC176x?  Is your fork different from the main one?  I have tried on my SKRV1.3 with no luck.  Compiles OK.

p3p, 2019-05-01:
yes it works for LPC176x, my fork is used by the Marlin LPC176x build, as long as the pin doesn't have any other filters, pullups etc on the board it should just work, I don't have a SKR 1.3 to test on, but on Re-ARM and SBase it works on quite a few pins.

doggyfan, 2019-05-01:
> 
> 
> yes it works for LPC176x, my fork is used by the Marlin LPC176x build, as long as the pin doesn't have any other filters, pullups etc on the board it should just work, I don't have a SKR 1.3 to test on, but on Re-ARM and SBase it works on quite a few pins.

oooo, I wonder what other other interesting differences/changes you've made p3p ?     anything nice ?

DimchikUA, 2019-05-01:
[image]
[Datasheet ws2812](https://static.chipdip.ru/lib/554/DOC001554925.pdf)
The datasheet for ws2812 says that the signal input should be 0.7 of the supply voltage.
Based on this 0.7 * 5V = 3.5V. It can be seen that 3.3v is not enough for the circuit to work. For everything to work, you can use the level converter, but you can cheat and lower the ribbon supply voltage a little bit, but then the brightness will be lower. Using a diode in the tape will provide a small drop in the supply voltage of the first LED and the circuit will work. The first LED will issue a control signal to the next diode with a voltage equal to its power supply. So everything will work.

I have not connected my strip yet. And I can not say that it works on pin P1_27. I looked at the scheme of SKR-1.3 and I think that P1_27 is not the best idea, since it has a RC filter.
[image]
[SKR-V1.3-SCH.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-SCH.pdf)
This will distort the outgoing signal. You need to either remove the capacitor and resistor (10K), or look for another pin. I will most likely use unoccupied pins with E1 plugs like EN/STP/Dir, or UART plugs from Z, E0 or E1. They are not yet occupied and do not have filters.

p3p, 2019-05-01:
Afraid I just add LPC176x support using my Arduino framework that Marlin on, I haven't pushed it upstream as my Arduino framework api hasnt stableised enough yet. (I probably could really but I control what libraries Marlin uses for LPC build so it doesn't matter)

@DimchikUA 
Practically speaking my Neopixel strips all work with an LPC176x 3.3v logic level with a clean 5.0v supply, Theoretically speaky as you said the spec says they shouldn't.

jmdearras, 2019-05-02:
> Afraid I just add LPC176x support using my Arduino framework that Marlin runs on, I haven't pushed it upstream as my Arduino framework api hasnt stableised enough yet. (I probably could really but I control what libraries Marlin uses for LPC build so it doesn't matter)
> 
Does this mean the main branch does not yet support the LPC176x?

p3p, 2019-05-02:
> Does this mean the main branch does not yet support the LPC176x?

Of the Neopixel library? yes it doesn't have LPC176x support I didn't push it upstream, but that isn't used by Marlin for LPC176x builds so it doesn't matter. Marlin bugfix-2.0.x has LPC176x support for neopixel without any modification .. other than enabling the feature and setting a appropriate pin for your board in the main config.

jmdearras, 2019-05-02:
> Of the Neopixel library? yes it doesn't have LPC176x support I didn't push it upstream, but that isn't used by Marlin for LPC176x builds so it doesn't matter. Marlin bugfix-2.0.x has LPC176x support [...]

OK, this stream has been a little confusing to me.  You are saying the Marlin 2.0.x main branch supports neopixels on the LPC1768 now?
Thanks.

codycato, 2019-05-26:
I'm new to coding etc.  But I'm getting Adafruit_Neopixel.h  directory or file cant be found. 

I have neopixel installed plus platformio...had everything working on1.1.9 arduino.  Help

SISLANGER, 2019-05-26:
@codycato in PlatformIO click on the alien in the first column somewhere I’d update library’s give that a go.

codycato, 2019-05-26:
Everything is up to date. I tried Pin p1_27 etc still no luck

SISLANGER, 2019-05-26:
@codycato Your in luck give me 5 minutes

SISLANGER, 2019-05-26:
@codycato Adafruit_Neopixel.h directory or file cant be found. this is down to platformio 
[image]

SISLANGER, 2019-05-26:
under project task in the left column at the very bottom will be update project libraries click on that then once that's done click on rebuild intelligence index

SISLANGER, 2019-05-26:
Once that's all ok
This is how I just got mine to work
https://www.makenvape.uk/3d-printing/3d-printing-guides/skr-v1-3-neopixel-bltouch/

codycato, 2019-05-26:
Found the problem...Adafruit_NeoPixel has to be under lib_deps inside platformio.ini

saranloyolite, 2019-09-01:
@SISLANGER 
I tried your guide but it's not working. I even tried removing the BLTouch and disabled it in Marlin. Still no luck!
Any advice? Thanks in advance!

SISLANGER, 2019-09-01:
@saranloyolite  what exactly is not working

saranloyolite, 2019-09-02:
> @saranloyolite what exactly is not working

Both the Neopixels and the Bl touch are not working.

I followed the instructions on your guide, wired up the Neopixel(ws2812b - 20 LEDs) to an external 5V PSU with common ground to the SKR and connected them to the servo pins. (double checked if 5V, ground, and signal are wired right and checked if the strip works with a nodeMCU).

Following are the code changes:
1. Configuration.h 
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB + NEO_KHZ800
  #define NEOPIXEL_PIN     P2_00
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 20
  #define NEOPIXEL_IS_SEQUENTIAL
  #define NEOPIXEL_BRIGHTNESS 200
  #define NEOPIXEL_STARTUP_TEST
  //#define NEOPIXEL_BKGD_LED_INDEX  0
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 }
#endif

2. Configuration_adv.h 
#define LED_CONTROL_MENU
  #if ENABLED(LED_CONTROL_MENU)
    #define LED_COLOR_PRESETS 
    #if ENABLED(LED_COLOR_PRESETS)
      #define LED_USER_PRESET_RED        255  
      #define LED_USER_PRESET_GREEN    255 
      #define LED_USER_PRESET_BLUE       255 
      #define LED_USER_PRESET_WHITE    255  
      #define LED_USER_PRESET_BRIGHTNESS 255  
      #define LED_USER_PRESET_STARTUP      
    #endif
  #endif

Connected the BL Touch to Zmax pins.

Following are the code changes:
1. pins_BIGTREE_SKR_V1.3h to:
#define SERVO0_PIN  P1_24

2. Congifuration.h
#define BLTOUCH

3. Configuration_adv.h
#if ENABLED(BLTOUCH)
  //#define BLTOUCH_FORCE_SW_MODE
  #define BLTOUCH_DELAY 200
  #define BLTOUCH_SET_5V_MODE
  //#define BLTOUCH_FORCE_MODE_SET
  //#define BLTOUCH_HS_MODE
  //#define BLTOUCH_LCD_VOLTAGE_MENU
#endif // BLTOUCH

Compiled and uploaded the firmware, and the BL touch is going on an erratic stow and deploy cycle and the neopixels don't light up even after I turned it on from the LCD menu.

PS: I am on the latest Marlin bugfix 2.0 (downloaded it on 2019 09 01)

And idea what might be going wrong?

SISLANGER, 2019-09-02:
@saranloyolite 

replace #define NEOPIXEL_TYPE NEO_GRB + NEO_KHZ800
with #define NEOPIXEL_TYPE NEO_GRB

If no luck with P1_24 some have had issues try
#define SERVO0_PIN  P1_26 and move cables to the Y Max

Disable below for the moment I found I didn't need it, and caused some issues.
//#define BLTOUCH_SET_5V_MODE

saranloyolite, 2019-09-02:
@SISLANGER 

I tried with #define NEOPIXEL_TYPE NEO_GRB at first but didn't work and after reading through the adafruit neopixel library I added the NEO_KHZ800. Doesn't seem to make any difference.

Shall try BL Touch with Y max pin with 5V mode undefined and get back to you. Hopefully, that would work. Even then I can't seem to understand why the neopixels are not working from PIN2_00.

SISLANGER, 2019-09-02:
neopixel library part is already done by Marlin so just the NEO_GRB is required.

Try disabling #define NEOPIXEL_IS_SEQUENTIAL 
I know It made me keep thinking it was not working when it was at least while testing.

Also here is my configuration.h part (striped down) see if you can spot things, ensure servo is on at bottom may help as well.

//#define RGB_LED
//#define RGBW_LED

#if EITHER(RGB_LED, RGBW_LED)
  //#define RGB_LED_R_PIN 34
  //#define RGB_LED_G_PIN 43
  //#define RGB_LED_B_PIN 35
  //#define RGB_LED_W_PIN -1
#endif

// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN    P2_00    // LED driving pin
  #define NEOPIXEL_PIXELS 77       // Number of LEDs in the strip
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 255  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
#define SERVO_DELAY { 300 }

saranloyolite, 2019-09-02:
@SISLANGER 

OMG! #define NUM_SERVOS 1 did the trick! Yaay! The neopixels are working now! Thanks a ton!

But the BL Touch is still randomly deploying and stowing!! Homing and bed leveling works, but it keeps randomly stowing and deploying! I am scared it will destroy the Bl Tocuh!! Any thoughts on this?

SISLANGER, 2019-09-02:
Double check the wiring, also pull out the switch cable from Y is it still doing it?

SISLANGER, 2019-09-02:
check the orientation of the GND SIGNAL IN THE Y MAX

saranloyolite, 2019-09-02:
The wiring is correct 5V, GND and SIGNAL. 
I tried with the X MAX and Z max pins as well and the same problem.
I was scared that mt BL Touch was damaged in some way so switched it back to the servo pin and the problem went away! 
This looks like some interference on the Endstop MAX pins!! :((

SISLANGER, 2019-09-02:
I had that at the very beginning, with one of the machines, trying to remember what it was.

saranloyolite, 2019-09-02:
Oh, okay that's a relief! Do let me know when you remember. Really appreciate all the support, man!

In the meanwhile I shall tinker around with it a bit more and see if I can sort this out! Shall post if it works. :)

SISLANGER, 2019-09-02:
I have you checked the Z-min cable orientation and set correctly in Marlin

SISLANGER, 2019-09-02:
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_ZMIN_PROBE
#endif

#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.

#define Z_MIN_PROBE_ENDSTOP_INVERTING true // set to true to invert the logic of the probe.

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define Z_MIN_PROBE_PIN P1_25 // Z_MIN_PIN

#define BLTOUCH

MAKE SURE 
#define Z_SAFE_HOMING is enabled

SISLANGER, 2019-09-02:
Sorry meant double check Y-Max orientation

saranloyolite, 2019-09-02:
> #define ENDSTOPPULLUPS
> #if DISABLED(ENDSTOPPULLUPS)
> // Disable ENDSTOPPULLUPS to set pullups individually
> //#define ENDSTOPPULLUP_XMAX
> //#define ENDSTOPPULLUP_YMAX
> //#define ENDSTOPPULLUP_ZMAX [...]

Yes, all of the Marlin configurations seems perfect!

I hadnt defined the #define Z_MIN_PROBE_PIN P1_25 // Z_MIN_PIN.
Defined it now and still the problem persists! :(

saranloyolite, 2019-09-02:
> Sorry meant double check Y-Max orientation

Y MAX orientation is also fine. I have cross-checked with the pinout diagram as well!

saranloyolite, 2019-09-02:
Also, this might be a stupid question, but I was going through this post where the ZMAX pins are proposed to be used to drive the Neopixels!
Link: https://reprap.org/forum/read.php?1,855543
Will this work?

SISLANGER, 2019-09-02:
Make sure with everything connected that you use the M502, followed by M500, followed by M501, then in menu reset BLTOUCH

SISLANGER, 2019-09-02:
Z_MAX was unreliable for me, colours wouldn't always match, never quite right for me.

saranloyolite, 2019-09-02:
> Make sure with everything connected that you use the M502, followed by M500, followed by M501, then in menu reset BLTOUCH

Did this, no luck still :(
Dont think this is going to work for me!

> Z_MAX was unreliable for me, colours wouldn't always match, never quite right for me.

The neopixels dont even light up for me when I connect to Z_MAX!!

jmdearras, 2019-09-02:
maybe I missed it, but the endstop pins on the SKR1.3 have a cap on the PCB
that must be removed for neopixels to work

SISLANGER, 2019-09-02:
@saranloyolite do you have #define SOFT_PWM_DITHER enabled if so disable that

SISLANGER, 2019-09-02:
@jmdearras ? CAP ?

saranloyolite, 2019-09-02:
> SOFT_PWM_DITHER

It is disabled!

> @jmdearras ? CAP ?

Capacitor

SISLANGER, 2019-09-02:
@saranloyolite #define BLTOUCH_SET_5V_MODE again see if that fixes issue.

saranloyolite, 2019-09-02:
> @saranloyolite #define BLTOUCH_SET_5V_MODE again see if that fixes issue.

I tried that already some time back. Makes no difference!

I was looking up how the TFT pins were used on the SKR v1.1 to drive the BL Touch. It seems to be tx Rx on the v1.3, not sure if I should try that! :/

SISLANGER, 2019-09-02:
@saranloyolite I think this is it
in pins_BIGTREE_SKR_V1.3.h

//
// Z Probe (when not Z_MIN_PIN)
//
//#ifndef Z_MIN_PROBE_PIN
  //#define Z_MIN_PROBE_PIN  P1_24
//#endif

saranloyolite, 2019-09-02:
> @saranloyolite I think this is it
> in pins_BIGTREE_SKR_V1.3.h
> 
> //
> // Z Probe (when not Z_MIN_PIN)
> //
> //#ifndef Z_MIN_PROBE_PIN
> //#define Z_MIN_PROBE_PIN P1_24
> //#endif

Tried, no luck! :(

SISLANGER, 2019-09-02:
Damn, I will keep thinking for you, let me know if you find a fix first

SISLANGER, 2019-09-02:
in platform.ino double check

[platformio] 
src_dir      = Marlin
build_dir    = .pioenvs
lib_dir      = .piolib
libdeps_dir  = .piolibdeps
boards_dir   = buildroot/share/PlatformIO/boards
env_default  = LPC1768

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
build_flags = -fmax-errors=5
  -g
  -ggdb
lib_deps =
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  LiquidCrystal@1.3.4
  TMCStepper@<1.0.0
  Adafruit NeoPixel@1.1.3
  https://github.com/lincomatic/LiquidTWI2/archive/30aa480.zip
  https://github.com/ameyer/Arduino-L6470/archive/dev.zip
  https://github.com/trinamic/TMC26XStepper/archive/c1921b4.zip
  https://github.com/mikeshub/SailfishLCD.git
  https://github.com/mikeshub/SailfishRGB_LED.git
  https://github.com/mikeshub/SlowSoftI2CMaster.git

[env:LPC1768]
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip
framework         = arduino
board             = nxp_lpc1768
build_flags       = -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin/src/HAL/HAL_LPC1768/include -IMarlin/src/HAL/HAL_LPC1768/u8g ${common.build_flags}
# debug options for backtrace
#  -funwind-tables
#  -mpoke-function-name
lib_ldf_mode      = off
lib_compat_mode   = strict
extra_scripts     = Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_LPC1768>
monitor_speed     = 250000
lib_deps          = Servo
  LiquidCrystal
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  TMCStepper@<1.0.0
  Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/master.zip
  https://github.com/mikeshub/SailfishLCD.git

saranloyolite, 2019-09-02:
Yes, I shall also try some things as well and let you know if anything works

Since the Z_MAX worked for you I am guessing it has to do with my BL Touch, which is a clone from Trianglelab. I can try with a friend's BL Touch and check. He is currently out of town and will be back day after.

saranloyolite, 2019-09-02:
My platformio.ini is slightly different :/

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = LPC1768

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5
  -g
  -ggdb
lib_deps =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal@1.3.4
  TMCStepper@<1.0.0
  Adafruit NeoPixel@1.1.3
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

[env:LPC1768]
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip
framework         = arduino
board             = nxp_lpc1768
build_flags       = -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin/src/HAL/HAL_LPC1768/include -IMarlin/src/HAL/HAL_LPC1768/u8g ${common.build_flags}
# debug options for backtrace
#  -funwind-tables
#  -mpoke-function-name
lib_ldf_mode      = off
lib_compat_mode   = strict
extra_scripts     = Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_LPC1768>
monitor_speed     = 250000
lib_deps          = Servo
  LiquidCrystal
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  TMCStepper@<1.0.0
  Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/master.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip

the first part - [platformio] is different

SISLANGER, 2019-09-02:
Mine is a clone from geeetech but I do remember someone else on FB having a problem with Trianglelab

what part is different

SISLANGER, 2019-09-02:
might just be version difference, it was working so didn't update any further as there was no need.

[5 more replies]