MarlinFirmware/Marlin #13639: Attached documentation for M425 and G425

marcio-ao, 2019-04-09:
I noticed that there wasn't any documentation for M425 and G425. I cloned https://github.com/MarlinFirmware/MarlinDocumentation/tree/master/_gcode and added some .md files in there, but it does not let me check them in.

So, given that I don't know what else to do, I am attaching the files here:

[G425.md.txt](https://github.com/MarlinFirmware/Marlin/files/3060815/G425.md.txt)
[M425.md.txt](https://github.com/MarlinFirmware/Marlin/files/3060817/M425.md.txt)

p3p, 2019-04-09:
You should be able to create a pull request on the https://github.com/MarlinFirmware/MarlinDocumentation repository the same way you usually would here.

marcio-ao, 2019-04-09:
@p3p: Ummm, okay. Now it worked. Must have been user error before. Thanks for the heads up.

closed 2019-04-09 by marcio-ao