MarlinFirmware/Marlin #13642: [BUG] (short description)

sonicalchemy, 2019-04-10:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

This is the newest 10 April 2019 marlin 2 bug fix. Getting this error message. 

In file included from sketch/MarlinConfig.h:30:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
Configuration.h:226: error: missing binary operator before token "("
 #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
           ^
In file included from sketch/MarlinConfig.h:30:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
Configuration.h:1035: error: missing binary operator before token "("
 #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
           ^
Configuration.h:1128: error: missing binary operator before token "("
 #if ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_BILINEAR, AUTO_BED_LEVELING_UBL)
        ^
Configuration.h:1154: error: missing binary operator before token "("
 #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
           ^
Configuration.h:1223: error: missing binary operator before token "("
 #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
           ^
In file included from sketch/MarlinConfig.h:30:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
Configuration.h:2064: error: missing binary operator before token "("
 #if EITHER(RGB_LED, RGBW_LED)
           ^
Configuration.h:2093: error: missing binary operator before token "("
 #if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
        ^
In file included from sketch/MarlinConfig.h:32:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
Configuration_adv.h:81: error: missing binary operator before token "("
   #if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
           ^
In file included from sketch/MarlinConfig.h:32:0,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
Configuration_adv.h:1053: error: missing binary operator before token "("
 #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
           ^
exit status 1
missing binary operator before token "("

[Marlin2TMCsmollyerror.zip](https://github.com/MarlinFirmware/Marlin/files/3063287/Marlin2TMCsmollyerror.zip)

reloxx13, 2019-04-10:
cannot reproduce. latest marlin compiles fine with ur provided config.

redownload marlin, ur marin source seems corrupted.

fix ur issue template. if u want others to bring effort to help u, u should bring the same effort.

marcio-ao, 2019-04-10:
@sonicalchemy: What are you using to compile? Arduino IDE? If so, what version?

sonicalchemy, 2019-04-10:
tried it in the previous version of Arduino IDE and list updated to 1.8.9 and tried again. Updated libraries as well. Redownloaded the Marlin 2 and tried that again and after all that still get the same issue. I took the config.h and adv after updating the settings for my printer and copied them into the Marlin file I have and no success. Is there a full version of the Marlin 2 now that I am missing in the file or am I just missing something. Please feel free to educate me if I am screwing up something. I know Marlin 2 is 32 bit enabled now so it is different to the 1.x branch. I am on a Mac and would love any hints. I realise this may not be a bug. Tell me here if it is not and i will copy and take it to the forum. Thanks for your help. I am feeling like a bit of a dunce at the moment. Feel pretty comfortable changing and working with the 1.x branch and want to start in trying out features of the 2 branch. Let me know if there is anything else you need to know from me. I am using an i3 clone and it has TMC2130 in Standalone and it has some custom pin assignments on the board from the last TMC2130 in Stealthchop mode as well as being set up for filament change and M600 for filament detection. Anything else I missed or you want to know. I think the rest you should be able to see in the Config.h.  All help and info is greatly appreciated.

closed 2019-04-10 by thinkyhead