MarlinFirmware/Marlin #13658: #define PIDTEMPBED    error printer halted when enable

bogaskilop, 2019-04-11:
I configured  marlin 2.0 for CR-10 test printer in my workshop  just for fun and to check if marlin 2.0 is usable now 

but when i enabled  BED PID : 

#define PIDTEMPBED  
when i try to heat up bed now i get this error :
  error printer halted

I wanted to enable this to make a new bed PID auto tune  but when enabled it just dont work 
when i  disable that it works and bed can be heated.

Did i forgot to change anything else in configuration.h  regarding bed pid autotune or there is an error in code  ?

Ludy87, 2019-04-11:
adopt the values from the previous version

Stady234, 2019-04-11:
Are you saying after enabling the PID for the BED you did a regular heating or the PID heating? and it made error...

bogaskilop, 2019-04-12:
@Stady234    when i enabled that option   i can not  heat up the bed   to make a pid tuning 

printer halted  that's all  even if you will try to set  temperature manually  it just stops with that error

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
If youre still having issues with Creality configs, feel free to use my branch for reference

https://github.com/InsanityAutomation/Marlin/tree/Creality_2.0_Bleeding