MarlinFirmware/Marlin #13662: [FR] (feature request title) Is it possible for Marlin to use duetwifi?

Msq001, 2019-04-12:
Esp3d is a compatible solution, but it's too slow to transfer gcode files with UART. Does anybody have any ideas?

Bob-the-Kuhn, 2019-04-12:
Short answer - Duet is not currently supported by Marlin. There is nothing in the hardware that would prevent someone from creating/modifying the needed files.

But I may not be answering your question.

Transfer speed optimization questions are probably best asked on the Duet forums.

Worst case is you take the SD card out of the Duet and plug it into your PC.

FYI - The Re-Arm LPC1798 board can be setup in Marlin to use a composite USB connection.  The on-board SD card is mapped as a virtual drive on the PC and the host software share the same physical USB connection.  Dropping a 8Mbyte GCODE file on the virtual disk took about 10 seconds to complete.  Uploading the same file via the host software (Repetier Host) takes multiple minutes.

Msq001, 2019-04-12:
Thank you for your reply.  Whether it is necessary to add the duetwifi-related API to marlin, because it will be a time-consuming task to start porting.

closed 2019-04-30 by Msq001