MarlinFirmware/Marlin #13666: LCD menu and Z offset not saving

bogaskilop, 2019-04-12:
when editing Z offset for bl touch  after setting up offset and choose  save settings  from lcd menu  settings are not saved   but   it changes and save   with M851  and m500 in terminal

InsanityAutomation, 2019-05-12:
Can you test with todays bugfix branch? A PR was just merged that may fix this. If not we may need to take a closer look. If this is now working, please go ahead and close the issue. Thanks.

bogaskilop, 2019-05-14:
will test  tomorrow on one printer i need to change few settings in firmware for new nozzle and different bl touch location so it is needed to put new firmware and what about my issue  #13773  ?

InsanityAutomation, 2019-05-14:
@bogaskilop I dont have a width sensor to do any testing on that one with. If I had hardware here, I would be able to debug it. Ive gotten a couple reports of issues there.

bogaskilop, 2019-05-14:
@InsanityAutomation ok do you know maybe on wich marlin version width sensor is working good?  i want to make few tests with this sensor and my printers.

boelle, 2019-07-21:
maybe configs would be usefull?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle