MarlinFirmware/Marlin #13667: [bugfix-2.0.x] compile error -lcdprint_u8g.cpp / U8GLIB_ST7920_128X64_RRD

lucagaff, 2019-04-12:
**Description**
Downloaded the latest bugfix-2.0.x code on 11 April 2019. Got my config files pasted in the marlin directory and tried to compile. I get the following output from the Arduino IDE:

```
_sketch\src\lcd\dogm\lcdprint_u8g.cpp: In function 'int lcd_put_wchar_max(uint32_t, pixel_len_t)':

sketch\src\lcd\dogm\lcdprint_u8g.cpp:36:24: error: 'class U8GLIB_ST7920_128X64_RRD' has no member named 'getPrintCol'

unsigned int x = u8g.getPrintCol(),

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp:39:23: error: 'ret' was not declared in this scope

u8g.setPrintPos(x + ret, y);

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp:39:28: error: 'y' was not declared in this scope

u8g.setPrintPos(x + ret, y);

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp: In function 'int lcd_put_u8str_max(const char*, pixel_len_t)':

sketch\src\lcd\dogm\lcdprint_u8g.cpp:45:24: error: 'class U8GLIB_ST7920_128X64_RRD' has no member named 'getPrintCol'

unsigned int x = u8g.getPrintCol(),

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp:48:23: error: 'ret' was not declared in this scope

u8g.setPrintPos(x + ret, y);

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp:48:28: error: 'y' was not declared in this scope

u8g.setPrintPos(x + ret, y);

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp: In function 'int lcd_put_u8str_max_P(const char*, pixel_len_t)':

sketch\src\lcd\dogm\lcdprint_u8g.cpp:53:24: error: 'class U8GLIB_ST7920_128X64_RRD' has no member named 'getPrintCol'

unsigned int x = u8g.getPrintCol(),

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp:56:23: error: 'ret' was not declared in this scope

u8g.setPrintPos(x + ret, y);

^

sketch\src\lcd\dogm\lcdprint_u8g.cpp:56:28: error: 'y' was not declared in this scope

u8g.setPrintPos(x + ret, y);

^

Multiple libraries were found for "U8glib.h"
Used: C:\Users\New User\Documents\Arduino\libraries\U8glib
Not used: C:\Users\New User\Documents\Arduino\libraries\arduino_224203
exit status 1
Error compiling for board Sanguino._
```

**Steps to Reproduce**
Load Marlin code in Arduino IDE
Click on Verify
Code does not compile
**Expected behavior:** [What you expect to happen]

Code should compile normally

**Actual behavior**: [What actually happens]

Error during compilation listed above.

**Additional Information**
U8Glib 1.19.1, also tried downgraded to 1.18.0 but had no different outcome.
Include a ZIP file containing your Configuration.h and Configuration_adv.h files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3073468/Marlin.zip)

Many thanks in advance to anyone able to help.

7eggert, 2019-04-16:
You've got the library installed twice.  Maybe that's the cause of your problem.

I did compile successfully with U8glib 1.19.1 using an official release of arduino (1.6.x and 1.8.8 both) and the skynet board definitions.

lucagaff, 2019-04-24:
Thanks for looking at this.
How do I delete the extra\redundant libraries?

jimmyeao, 2019-04-26:
Go to your My Docs folder, Arduino, Libraries and delete it from there

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle