MarlinFirmware/Marlin #13668: BLTouch Calibration Issue

sgsalman, 2019-04-12:
Hello guys, Firstly I want to say, my language not very well, So I couldn't find my problem's solution in the deeply. Im waiting a little help.

I am using Ender3-Pro and TH3D Unified Firmware (U1.R2.10a) (actually this is best firmware or not, I dont know). I have installed BLTouch and set the required settings from firmware. 
Everything ok but I got a problem with BLTouch. When it is auto homing Z axis, probe's pin touches the bed but not suddenly stop. It continues to move toward the Z(-) axis. As such, the nozzle pushes the bed a little bit. I watch too many videos and In all videos, the probe's pin stops suddenly when it touch the bed. Probably I have an firmware issue but i couldnt find it. 

**Config files**
(https://github.com/MarlinFirmware/Marlin/files/3073137/Configuration_sgsalman.zip)

**Video**
(https://photos.google.com/share/AF1QipN7G7LdfAiukdPC3aVkaLqykz1_MpQz6DNRcA3NV-oE4ZYdPMjabNfnnH1G9JdE0w?key=R0tUM3kxaC0xVVRsYmxuY1hUYXBlY0k4aGs4UTZR)

FanDjango, 2019-04-12:
If your BLTouch is a V3, consider reading this one #13345

sgsalman, 2019-04-12:
@FanDjango It says for Creality V1 Mainboard in guidebook

sgsalman, 2019-04-12:
@FanDjango post is so complicated :( I have not well english. Can you please copy paste the certain solution? So I can try. pleaseee

FanDjango, 2019-04-12:
> Can you please copy paste the certain solution? 

Nothing is certain. 

Is the BLTouch connected (cables, pins, configuration in firmware) correctly?

But first, is it a V3 BLTouch?
If yes, to avoid needing a firmware update, try the solution of connecting a small 1K resistor such as described in the other issue - there are is a good picture where one person did it very easily.

Test it with your fingers pushing the pin and do not let the nozzle crash.

FanDjango, 2019-04-12:
I got the two pics for you. Good Luck
[image]
[image]

sgsalman, 2019-04-12:
@FanDjango thank you so much. I understood from the topic that u gave me but I couldn't realize the which wire was C7 :D thnkx a lot.

sgsalman, 2019-04-12:
1. Remove the capacitor(recommended)

          Even if Smart V3.0 produced since April 5th, 2019, we still recommend removing capacitors.

It is a safe and complete solution.  Removing it does not cause any problems in your system. 

Moreover, also any future use of any sensor will provide better precision. 

It works well in Logic voltage Free mode (default : open drain).

ex) C7(Melzi/Ender-3) / C45(Anet)

from https://www.antclabs.com/wiring3-1

Can I trust this recommended solution? It is more easy for me ?

FanDjango, 2019-04-12:
If you can locate the capacitor and you can work with a good very small soldering iron to desolder the SMD component without damaging the circuit board, you can try that. If you are that good at desoldering, you can probably also solder it back on some day if you want to. Just be careful. 
More tricky is to use a cutter or force - so easy to damage copper traces on the board. Your choice.

I do believe that solution will work but that's not the point.

Because many people prefer NOT to modify their hardware (controllerboard). Understandable, easier to ADD something (1K resistor) in the wiring than to REMOVE something on the controller.

Or, even better, not modifying anything and just updating firmware.

sgsalman, 2019-04-12:
When you say like that, it is seem to my eye easier :D okey I'will try to adding 1k resistor. but I have 1 more question. I plugged the wires according to guidebook. 
so in pin27 board

yellow : SIG 
blue    : GND
red     : VCC

Is that correct? or will I need to change blue and red ?

FanDjango, 2019-04-12:
You can take it in two halves:
Part 1: Giving the BLTouch POWER and SERVO-SIGNAL - that's the one cable: Brown = GND RED = VCC, so it has power, then YELLOW = SIGNAL to give it commands.
Part 2: Coming back out, BLACK is GND (again, and some people leave it hanging and don't use it, because it is the same as Brown), and WHITE = SWITCH-SIGNAL-OUTPUT
Not sure why you see "Blue"???

sgsalman, 2019-04-13:
yeah there is no BROWN in my bltouch smart v3 pre april application :D there is blue cable instead of brown as I said previous comment.
yellow : SIG
blue : GND
red : VCC 
I checked two times, because of I suspected myself suspect :D

boelle, 2019-07-21:
is this a bug or question?

boelle, 2019-10-24:
you better ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs and fr's only

closed 2019-10-24 by boelle