MarlinFirmware/Marlin #13670: SKEW_CORRECTION Formula

MauricioPG, 2019-04-12:
### Description

I´ve adjusted the 1.1.9 Marlin Firmware into a HICTOP 3DP-08.

Calibrating the printer, I´ve tried the SKEW-CORRECTION, cause I have a YZ skew, about 4 mm / 10mm.

Printed the YACS, mesures, etc., inserted values and nothing.

Well, came to configuration.h, read again everything and used the given calculation formulas

* - Compute AB     : SQRT(2*AC*AC+2*BD*BD-4*AD*AD)/2
*  - XY_SKEW_FACTOR : TAN(PI/2-ACOS((AC*AC-AB*AB-AD*AD)/(2*AB*AD)))

to manually verify the values. Applying the second formula in EXCEL, with the values given for a perfect 200 mm square (as of configuration example), the result is -0,377968616, where it should be 0.0.

I´ve combed all codes trying to locate the actual used calculations for SKEW, but didn´t find. And I have some experience in programming...

So, can you guys check it out? Maybe is just correct the explanation on configuration.h.

Or, am I getting something wrong?

boelle, 2019-07-21:
is this a bug or question?

boelle, 2019-10-24:
you better ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

this list is for bugs and fr's only

closed 2019-10-24 by boelle