MarlinFirmware/Marlin #13672: [BUG] UBL not probing correct number of points

mtwomey, 2019-04-12:
### Description

I am not 100% certain this is a bug, but I've checked everything I can think of and it _seems_ like it may be a bug. On the bugfix release (or the 1.1.9 release), when I setup UBL and define the GRID_MAX_POINTS values, it seems to only probe a grid that is "x minus 2" of the specified grid size. For example with these settings:

```
#define MESH_INSET 0       
#define GRID_MAX_POINTS_X 3
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```
It probes  only a single point (in the middle of the bed).

With these settings:

```
#define MESH_INSET 0       
#define GRID_MAX_POINTS_X 5
#define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```
It probes 9 points (3 x 3)(The grid it produces is 5 x 5, but it only actually probes a 3x3 area)

### Steps to Reproduce

1. Setup UBL on Marlin 1.1.9 or the 1.1-bugfix release
2. Define GRID_MAX_POINTS values
3. Initiate G29 P1

**Expected behavior:** [What you expect to happen]

Probe X * Y number of points

**Actual behavior:** [What actually happens]

Probe (X-2) * (Y-2) number of points.

#### Additional Information

Example of results when setting GRID_MAX_POINTS_X 3:
```
    (  0,212)      (212,212)
        0       1       2
 2 |   .       .       .
   |
 1 |   .    [+0.019]   .
   |
 0 |   .       .       .
        0       1       2
    (  0,  0)      (212,  0)
```

gloomyandy, 2019-04-12:
Which bugfix release did you try (1.1.x or 2.0.x)? What sort of printer do you have? What type of probe are you using? 

What are you probe offsets and what is the setting for MIN_PROBE_EDGE? It is very likely that one or more of those will stop you from probing points around the very edge of the bed. So basically what you are seeing is normal and not a bug. 

Oh and please when reporting issues follow the request to upload your configuration files. it really does save a lot of wasted effort. For instance I could have checked the above values rather than having to ask you for them and I would also know what sort of system you have - deltas work very differently when it comes to probing - rather than having to assume it is some sort of Cartesian system.

mtwomey, 2019-04-12:
Yes, my apologizes on not attaching the config files. I've attached them here. I was using the 1.1 X bugfix as well as the 1.1.9 release. I'm using a Makerfarm Prusa i3v printer with a BLTouch 3.0.

I was _just now_ experimenting with the  **MIN_PROBE_EDGE** setting. I see what you're saying, this did make a different. When I set it to zero, I got 4 points probed (it was previously set to 20):

```
   (  0,212)      (212,212)
       0       1       2
2 |   .     +0.392  +0.986
  |
1 |   .    [-0.001] +0.231
  |
0 |   .       .       .
       0       1       2
   (  0,  0)      (212,  0)
```
Based on your explanation, I can now understand how this could be the intended functionality. My probe is "forward and to the right" of the origin in that graph - so it knows it can't reach the bottom and left edges.

What I'm hoping for - is for it to probe the "extents" of what it _can_ reach however. Is this possible?

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3074837/config.zip)

Roxy-3D, 2019-04-12:
UBL is trying to give you the full print envelope of your bed.   The problem is because of the probe offset, you can't probe every where you can print.   You can add an INSET to limit some of the print area.    But really...   You are better off manually filling in the unprobed parts of the mesh.

You can use G29 P3 to fill rows with 'reasonable' values.     And then edit those mesh points up or down a little bit if you see a need.   (The nozzle is a little too close...  or a little too far from the bed as it is printing the first layer.    You can also use G26 to print a mesh validation pattern that will help you know how well each mesh point reflects reality.)

gloomyandy, 2019-04-12:
As above you can use mesh inset to move the probe points, but probably a better option is to use more probe points that way you will still lose the edges but will be probing more of the bed. Then you can just use the probed values to perform an auto fill of the missing points.

mtwomey, 2019-04-12:
Ok makes sense - thanks folks. I was only using 3x3 right now until I got things tuned and was then going to increase it to something more. So I'll try that and do the mesh edit for the edges if needed.

closed 2019-04-12 by mtwomey