MarlinFirmware/Marlin #13677: [FR] Improvement proposal: Suspend and save the SD card

maxlinux2000, 2019-04-13:
Imagine you are printing something big that you need many hours to finish.
But you have to leave out and you don't want to leave the printer without control (for possible fires) or if you don't want just to sleep but the noise that the printer makes prevents you from doing so.

In these cases (and many many more) it would be very useful to be able to store the nozzle position on the SD card and be able to turn off the printer without having to start from scratch.

In my opinion, the system would have to work like this:
in the menu there must be an entry that says something like:

"Suspend and save nozzle position on SD card"
"Are you sure?" Yes / Cancel

When it is confirmed, the printer follow to finish print  the gcode code line where it is working and then saves in the SD an empty file with the name of the file +  the line number where it has contunuar + the height of the Z axis.
For example:
testcube.gcode_3398_Z3.75.resume
what does it mean:
<filename> _ <line number> _ <height Z axis> .resume

Once the file is saved in the SD card, the system homing the XY axes and leaving the Z axis as it is.

When we turn on the 3d printer again, Marlin goes to find the * .resume file and asks for confirmation to resume printing exactly from the point where it was suspended.

Obviously this applies only to the filaments that do not need a warm bed, but the PLA and the TPU and several others do not need it and this would be very useful.

You could compare this system with the "suspend to disk" of linux and windows.

swilkens, 2019-04-13:
How would you handle homing the axes before you resume printing? Presumably standard homing would interfere with the half-finished model already on the printbed?

comps, 2019-04-13:
You could assume (as mentioned by the OP) that the Z axis doesn't move between the "pause" and "resume", homing just X and Y, as is the case for many i3-style printers. It is likely that you would lose microstepping accuracy, but ie. with 2mm leadscrew pitch, you get a fullstep every 0.01mm, so +- 0.01mm on ie. 0.2mm layers is not a dealbreaker, especially if you can control where on the model it happens.

maxlinux2000, 2019-04-13:
There is another way to make the nozzle homing without the partially printed part bothering you.
Although you have to modify the 3D printer a bit. It involves adding to the Z axis an endstop above the axis.
In this way instead of doing the homing to low in the hot bed, it will be done up.
Thus the partially printed part will not disturb the nozzle homing.

This means that Marlin will have to know in the configuration the exact measurement of the Z axis.
We will have to create another script that makes the botton-top homing so that Marlin can measure the exact height of Z.

But this script will be very useful also for the Power_loss_recovery, because instead of doing the homing to low where there are partially printed pieces, it will do it up where the nozzle can not hit anything.

Of course it means adding an endstop on the Z axis, but our 3D printers already have a lot of loose wires :-)

AnHardt, 2019-04-13:
Is there any conceptual advantage over a (working) Power Loss Recovery?

InsanityAutomation, 2019-04-13:
@AnHardt Id think it would be lighter weight and less aggressive writing to the cards and therefore less prone to destroy them as power loss recovery defeats the wear leveling functions. Should be more reliable as it would be more of a controlled process as well. That said im not sure those gains are enough to justify the effort at this time. There are many other higher priority tasks.

comps, 2019-04-13:
Additionally, one would theoretically have greater control at which point to issue the pause, so ie. a full layer could be finished, the hotend could safely cool down before poweroff, the nozzle would clear the object (and not burn a hole in it), etc.
Sort of like a scheduled M600 combined with power loss recovery metadata.

maxlinux2000, 2019-04-13:
well, it could be done in parts.
First, you can save the position on the SD card so that you can resume by homing only the XY axis, without touching the Z axis.
It will not work well for everyone, but for those who have the 8mm metric threaded rod, it will surely work properly.

Later on, you can see how to develop the TopHoming necessary for the TR8x2 and TR8x8 strands and also to improve the Power_loss_recovery that currently works only partially (it can only be resumed once)

robbycandra, 2019-04-14:
My printer use Z-Max Homing... Therefore i just Pause Print and cut the power.

maxlinux2000, 2019-04-16:
There is another very important reason to implement this new feature as soon as possible.
In the underdeveloped countries they are very interested in 3D printing, but they can not print things, because their houses have only a few solar panels and the lucky ones a wind turbine.
This means that during the night or when there are clouds, they can not print and that is why they do not start, because they do not know how long the good weather will last.
It is possible that many of us have fun to print test packs that last 30 minutes, but there are others that would use the 3D printer to make a living.
Therefore, a system of "Suspend to SD" integrated in Marlin, would be THE SOLUTION to this class of problems.

regards

LinoBarreca, 2020-01-08:
While this could be implemented it can't work for several reasons:
1) you can't keep Z when you power off the stepper
2) when you switch off the printer the heated bed cools (and the piece pops off)
3) piece shrinkage (the piece shrinks as it cools down) even if you could restart the print without the first two problems you would clearly see the point where you paused it.
4) part rigidity would be shit in the first resumed layer (hot plastic to very cold plastic)

This could work only if the pause is very small in time.

closed 2020-04-29 by thinkyhead