MarlinFirmware/Marlin #13679: [BUG] (short description)

dmamontov, 2019-04-13:
sketch\src\sd\SdBaseFile.cpp: In member function 'sync':

sketch\src\sd\SdBaseFile.cpp:1448:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\dmamo\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/dmamo/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Ошибка компиляции для платы Arduino/Genuino Mega or Mega 2560.

dmamontov, 2019-04-13:
Anycubic Kossel config. Help

Bob-the-Kuhn, 2019-04-13:
>  internal compiler error: Segmentation fault

That's a tool chain problem.  I saw it recently when using PlatformIO to compile.  It stayed until I rebooted my computer.

paulgavrikov, 2019-06-16:
@dmamontov I had the same issue when compiling with Arduino IDE. Switching to PlatformIO did the trick for me.

closed 2019-06-17 by dmamontov

dmamontov, 2019-06-17:
just moved the folder to another place!