MarlinFirmware/Marlin #13683: Unkown Command :"" [BUG] (short description)

Mohamed-Ateef, 2019-04-13:
Hello , i was working First time on marlin and i upload code to my 3d printer on MKS GEN V1.4 
and Work Fine . 

One time endstop made short My bad .. then i trid the printer after that all thing work fine .. 
Saddly i was not able to upload new code .. until i get the resolve to change Name Of board many time until i upload code .. so there is something happen to my board ?! 
the big problem i can`t move axis Or anything from PC .. connect it By USB .. i got in Printer Log :
Unknown Command : echo:Unknown command: "G	0"
and i press move axis : 
echo:Unknown command: "E1"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "A"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
and its keeping print this line on serial monitor ! 
echo:Unknown command: ""

Bob-the-Kuhn, 2019-04-13:
Please ZIP up the following and drop it on your reply:
- The file **configuration.h**
- The file **configuration_adv.h**
- A copy of the log from power up through the start of the error message.

What host software are you using?

Mohamed-Ateef, 2019-04-14:
Hello Good Morning :) 
The Printer Log is : 
Connecting...
start
Printer is now online.
echo:Marlin 1.1.9
echo: Last Updated: 2018-08-01 | Author: (Mohamed,Atef,9/2/2019, default config)
echo:Compiled: Apr 13 2019
echo: Free Memory: 2914  PlannerBufferBytes: 1264
echo:Hardcoded Default Settings Loaded
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X79.88 Y79.88 Z1629.00 E99.76
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z3.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X2000 Y2000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1200.00 R3000.00 T2000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X5.00 Y5.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P97.78 I13.26 D180.23
echo:Z-Probe Offset (mm):
echo:  M851 Z-3.50
echo:Stepper driver current:
echo:  M906 X750 Y750 Z720
echo:Unknown command: "*‚ª"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "M1"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "E1"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "E11"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "G0"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "E11"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "A0"
echo:Unknown command: "E"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "A"
echo:Unknown command: "E"
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: ""
echo:Unknown command: "I00"
echo:Unknown command: "A0"
echo:Unknown command: "E1"
echo:Unknown command: ""
this Echo happen When i press move axis Any axis .. Expat this its always print in serial Monitor 
"echo:Unknown command: """
im using pronterface software

Marlin Config.H/ADV
https://drive.google.com/open?id=18v2zUgwSiXgdgh7DkVnUg7FU0bIGeT7E

Bob-the-Kuhn, 2019-04-14:
Sorry - I was not able to reproduce your problem.

I downloaded your files, disabled the TMC monitor and downloaded it to a 2560/RAMPS system.  I then connected via Pronterface.  Everything is the same in the log through` echo: M906 X750 Y750 Z720`.  I clicked on the direction arrows and M114 returned the expected results.

It looks like your system can hear the controller just fine but the controller has problems hearing Pronterface.

Try the following:
- Lower the baud rate
- Different USB cable
- Different USB port on your PC
- Repetier Host instead of Pronterface
- Remove & reinstall Pronterface

Mohamed-Ateef, 2019-04-14:
Hello , 
Nice to Hear From u Again :) 
So the board is Good well Nothing Destroyed ?! .. My Board is MKS GEN V1.4 .. 
i think that happen to me when i plugin the Fan on D1 on Board Maybe this is the problem ?! or not
i did : 
* Changed The USB Cable .
* Changed the Virson Of Proterface to new One .
* Changed the USB Port On PC . 
i Need to Be sure the board not have issue :( 
i Checked the board nothing bad happen to board nothing fire

Bob-the-Kuhn, 2019-04-14:
Did you try lowering the baud rate?

Mohamed-Ateef, 2019-04-14:
> Did you try lowering the baud rate?

No .. i didn`t change it last Night ! 
My Board Work Fine ?! or its have issue ?! 
Maybe because i connect fan to D1 ?!

Bob-the-Kuhn, 2019-04-14:
It definitely has a communications problem.  Going to 115200 or 57600 may be enough to make it happy.  

Connecting the fan to D1 shouldn't cause a problem unless the other side of the fan was tied to a voltage greater than 7 Volts.  If you apply too much voltage to the CPU then all kinds of strange things can happen.

Mohamed-Ateef, 2019-04-14:
> It definitely has a communications problem. Going to 115200 or 57600 may be enough to make it happy.
>
> Connecting the fan to D1 shouldn't cause a problem unless the other side of the fan was tied to [...]

if i change the baud rate stay same ? what will happen ?
the printer work fine stand alone none computer !

Bob-the-Kuhn, 2019-04-14:
If you lower the baud rate the only thing you'll notice is slower uploads of large Gcode files.

Mohamed-Ateef, 2019-04-14:
> If you lower the baud rate the only thing you'll notice is slower uploads of large Gcode files.

i Men .. if i lower Baud rate ! and problem still Have same issue what will be after that ?!

Bob-the-Kuhn, 2019-04-14:
If lowering the baud rate doesn't help try moving the printer and the PC closer so that a shorter USB cable can be used.

If neither of those work then it's time to replace the board.

Mohamed-Ateef, 2019-04-14:
> If lowering the baud rate doesn't help try moving the printer and the PC closer so that a shorter USB cable can be used.
> 
> If neither of those work then it's time to replace the board.

Ok Thanks , i will let u know what happen .. when i get home cuz im in work

Mohamed-Ateef, 2019-04-14:
> Did you try lowering the baud rate?

Hello again :) 
i was trying to lower Baud rate 
i Install New windows For that maybe the problem was with windows but not i think .
My board Not able to upload Code any more i think ! :( 
there anything i can check ?! can i upload Photo to show u it

closed 2019-04-16 by thinkyhead