MarlinFirmware/Marlin #13686: [QUESTION] On contribute pull request, how to efficiently update ALL example configurations?

FanDjango, 2019-04-14:
I would like to contribute. Need to add something in ALL example configuration.h files. Before tedious manual editing, anyone have a script to automate this?

swilkens, 2019-04-14:
The quickest way is probably to to the edits on a local computer and push the changes to Github, I don't know of a straightforward way to make multiple changes to multiple files in one commit using the web interface.

FanDjango, 2019-04-14:
Ok, so I will use TextCrawler ( = MASS FIND AND REPLACE Tool )

closed 2019-04-14 by FanDjango

Roxy-3D, 2019-04-14:
What I find to work well is Astro-Grep on Windows.   I can search for the text right next to where a new option is being added to the Configuration.h files.    The program finds it in all the Configuration.h files.   I high light the entire set, and tell it to Open those files.   

It will pop up 50 (or so) new edit windows with the cursor positioned exactly where it should be.   And then I just paste in the new text, save and exit each window.

Bob-the-Kuhn, 2019-04-14:
Find an editor that will do replaces on multiple files.  [Notepad++](https://notepad-plus-plus.org/download/v7.6.6.html) is one.

First I set Notepad++ as the default editor for all the file types used in C & C++
[image]

I use Astro-Grep on Windows to search for configuration.h files or for the desired text.  I then select all the files , right click and select **Open File**.  All the files are then automatically opened in Notepad++.

In Notepad++ I select the area of interest and then click the replace icon/tool.  I then make my edits.  Then I select **ALL** the replacement area, do a CNTRL C, go to the replacement dialog, clear out the **Replace with:** box and then paste in the new text.  Then you click the **Replace All in All Opened Documents** button.  

Take note of the number of replacements it made.  If that isn't the same as the number of files found by Astro-Grep then it gets tedious.  One by one select the files in NOTEPAD++.  If it's been changed then save & close it. When you find one that hasn't changed, edit it, save it & then close it.

As a double check look at the changes list in Github Desktop anbd verify the number of changed files is correct.

FanDjango, 2019-04-14:
Ok, got it.
Been racking my brains for a better solution but sometimes there is no alternative to good fast manual editing aided by some useful tools.
Kind of motivates one to avoid PRs that would involve new #defines in configuration'.h, though.
Thanks for the pointers.
Currently I am using solely TextCrawler, it will do the finding and the editing in one go.
I would not be surprised, however, on a meticulous search, to find unexplained differences in the example configs - especially if PR's span the creation of a new example config, as is happening right now. Is why I am pulling back my PR right now until I see how this plays out.
Another problem is my inexperience in operating GitHub, I come from a big-iron mainframe background and am used to slightly more rigid systems and different terminologies. But I am getting the hang of it now as long as I can close my own tentative PR's that I don't like the look of...
Somehow I kind of like the idea of building my own fork and keeping it going - safer than putting things into the upstream. But that is not the point of collaboration, is it?